raspberrypi/documentation #353: removing lightdm service command is incorrect

marvado69, 2016-05-01:
the command `update-rc.d lightdm remove` is incorrect, should be `update-rc.d -f lightdm remove`

XECDesign, 2016-05-04:
What page is this on?

marvado69, 2016-05-05:
Page: https://www.raspberrypi.org/documentation/remote-access/vnc/
To remove the service lightdm the -f option should be added in the command.

XECDesign, 2016-05-05:
I wonder if the correct thing to do is actually `sudo systemctl set-default multi-user.target`. Is the idea to prevent a 'real' X session from running and eating up resources? If so, that would be the correct systemd way to do it, I think.

marvado69, 2016-05-05:
Well, could be. Can't say because lack of knowledge.
Without the -f option it won't work.

XECDesign, 2016-05-05:
Does it prevent VNC from running or just doesn't prevent lightdm from starting on `:0`? Could you give `sudo systemctl set-default multi-user.target` a go and see if that works as expected?

marvado69, 2016-05-05:
It prevents lightdm from starting on `:0`.
I've restored the situation by command 'sudo update-rc.d lightdm defaults' and rebooted. Lightdm was on  ':0' and VNC on ':1'.
Then command 'sudo systemctl set-default multi-user.target' was given. After reboot the result was the same; Lightdm was on  ':0' and VNC on ':1'.

XECDesign, 2016-05-05:
Thanks. To keep things consistent and backwards compatible, I'll update the page using your original suggestion. It's probably worth revisiting that page at some point and properly re-writing it for jessie.

commit 30964158 (2016-05-05): Properly remove lightdm service in VNC instructions, close #353
closed 2016-05-05 by XECDesign