raspberrypi/documentation #358: Documentation about Kelvin values for AWB options available?

mrcg, 2016-05-17:
I would need table showing which kelvin values are used for each individual AWB option (e.g. sun, cloudshade, tungsten, fluorescent, etc.).

E.g.:
sun => 5000K
fluorescent => 3500K
flash => 6000K

bennuttall, 2016-05-17:
I was wondering this recently. Dived into @waveform80's [picamera](https://github.com/waveform80/picamera/) source but that was just referring to named AWB constants in the firmware. Perhaps @6by9 will know?

waveform80, 2016-05-17:
Yeah, I'm afraid picamera is nothing more than a (historically slim, these days slightly fatter) wrapper over the MMAL API so I can't shed much light on what kelvin values the settings might map to. @6by9, Naush (who's been working on the firmware for the V2 module - sorry I don't know his GitHub handle, but he's active on the [camera board](https://www.raspberrypi.org/forums/viewforum.php?f=43) as naushir) or @JamesH65 are probably your best bet for this sort of info.

6by9, 2016-05-17:
I don't know how accurately the AWB tuning has been done on either OV5647 or IMX219 sensor. 
The only info I can see in the firmware source is 
   DAYLIGHT,      /* 5000K - 6500K  */
   CLOUDY,        /* 6500K - 12000K */
   TUNGSTEN,      /* 2500K - 3500K  */
   FLUORESCENT,   /* 2500K - 4500K  */
There are no comments in the tuning as to the exact temperature used within those ranges.

JamesH65, 2017-03-02:
Without knowing what was used in the tuning the best we can do is the numbers 6x9 has posted. I doubt we will ever know for the v1 camera, but I'll update the docs with the table above.

JamesH65, 2017-03-08:
Fix merged, closing.

closed 2017-03-08 by JamesH65