raspberrypi/documentation #359: Two different versions of dt-blob.dts in repository, Google search points to out of date one

mtancret, 2016-05-18:
When I did a google search for "dt-blob.dts" the first link I get is this out of date version:
https://github.com/raspberrypi/documentation/blob/master/configuration/images/dt-blob.dts

It prevented my B+ with a firmware from Nov 18 2015 15:43:45 from booting.

After reading some forum posts, it appears that the up to date version is located at:
https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts

lurch, 2016-05-18:
When https://github.com/raspberrypi/documentation/blob/master/configuration/pin-configuration.md was first written (which is the page that links to https://github.com/raspberrypi/documentation/blob/master/configuration/images/dt-blob.dts) I believe the dts-blob wasn't included in the firmware repo. But now that it is, I guess somebody (@pelwell seems to be the DT expert?) ought to update the first page to suit.

pelwell, 2016-05-18:
The dt-blob uses the syntax of Device Tree but not the semantics - it isn't really Device Tree at all.

Keeping one dt-blob.dts up-to-date is hard enough - there really shouldn't be two. Let's delete the one in the documentation and replace any links with a reference to the firmware repo.

JamesH65, 2017-03-02:
I'm inclined to remove the sample section from the documentation altogether, get round out of date examples and links completely. Happy? If so I'll remove and close the issue.

6by9, 2017-03-06:
> I'm inclined to remove the sample section from the documentation altogether, get round out of date examples and links completely. Happy? If so I'll remove and close the issue.

I'd link to https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts as the master public blob.
Dom does have magic scripts that should run on every release to prompt him when there are changes on the vc4 tree to be reflected in that copy. Keeping it all in the firmware repo should mean that is consistent.

Certainly https://github.com/raspberrypi/documentation/blob/master/configuration/images/dt-blob.dts and any references to it need to go.

JamesH65, 2017-03-08:
fixed merged.

closed 2017-03-08 by JamesH65