raspberrypi/documentation #363: Kernel level debugging page

MartyMacGyver, 2016-05-20:
The kernel build instructions are found in the following tree:
https://www.raspberrypi.org/documentation/linux/kernel/

It would be helpful to have a page on how to configure and build the kernel and modules for debugging, to assist advanced users and contributors with working on code and/or better reporting errors, memory leaks, etc. on the Pi platform.

As a start, I've gathered some information on the settings to enable kernel memory leak debugging below, but there are other common settings and techniques I've not covered (e.g., building so that addr2line actually works correctly).

For memory leaks - before the build:

    # Customize the config to configure kernel memory leak debugging:
    # Either edit `.config` directly or...
    make menuconfig  # Menu driven, uses .config as a starting point
        # Kernel memory leak debugging:
            # Kernel hacking ? Memory Debugging
            CONFIG_SLUB_DEBUG_ON=y
            CONFIG_SLUB_STATS=y
            CONFIG_DEBUG_KMEMLEAK=y
            CONFIG_DEBUG_KMEMLEAK_EARLY_LOG_SIZE=4096
            CONFIG_DEBUG_KMEMLEAK_TEST=m
        # Save your changes before exiting!

After install and reboot:

    # Using kernel memory leak debugging
    # sudo mount -t debugfs nodev /sys/kernel/debug  # superfluous - already mounted
    sudo bash -c "echo scan > /sys/kernel/debug/kmemleak"
    sudo cat /sys/kernel/debug/kmemleak  # Repeat to see updated info
    
    # IF the `echo scan...` part above *fails*, check /var/log/kern.log
    # What I saw: "Kernel memory leak detector disabled",
    # "Early log buffer exceeded (####), please increase DEBUG_KMEMLEAK_EARLY_LOG_SIZE"
    # I increased CONFIG_DEBUG_KMEMLEAK_EARLY_LOG_SIZE and rebuilt/reinstalled the kernel
    
    # Another handy command for memory leak detection:
    sudo slabtop -sc

JamesH65, 2017-03-06:
I reckon this is a pretty good idea, would you be able to write a new page and submit a pull request?

JamesH65, 2017-06-06:
No comment from OP, and I don't have the time or knowledge to do this. If anyone fancies it, please send in a PR. For the moment. closing this issue.

closed 2017-06-06 by JamesH65