raspberrypi/documentation #372: ..

ntewinkel, 2016-06-03:


closed 2016-06-03 by ntewinkel

pelwell, 2016-06-03:
Now you know what that button does.

ntewinkel, 2016-06-03:
LOL. I thought I was in a terminal window when I typed "cd .." and hit return. oops!

bennuttall, 2016-06-03:
I know your game...

[image]

ntewinkel, 2016-06-03:
I don't know what "game" you're referring to, bennuttall?  Those 829 contributions are not raspberry pi related (they are client work for private repositories).
Is there a way I can delete this issue?

pelwell, 2016-06-03:
We can't, but don't worry about it - there was an invisible smiley at the end of Ben's message. He was probably suggesting that you were trying to up your reputation by being seen to contribute to our repo, but knowing the kind of people who write our documentation it's not a club you'd want to join. (BTW 829 is the number of people who have starred our documentation repo.)

lurch, 2016-06-03:
> but knowing the kind of people who write our documentation it's not a club you'd want to join

lolz

bennuttall, 2016-06-03:
> don't worry about it - there was an invisible smiley at the end of Ben's message. He was probably suggesting that you were trying to up your reputation by being seen to contribute to our repo

Indeed. Just kidding.

> but knowing the kind of people who write our documentation it's not a club you'd want to join.

Good point. Maybe we need a logo...

[image]

lurch, 2016-06-03:
The first rule of Documentation Club...

bennuttall, 2016-06-03:
...is to document the club's rules.

The second rule of Documentation club is ** to document the club's rules**.