raspberrypi/documentation #380: Consider using Etcher software for user friendly SD card prep

saildata, 2016-06-21:
I did a quick search and didn't see any duplicates on this - I apologize if it's already been discussed. The new cross-platform SD card software, [Etcher](www.etcher.io), makes it very easy to transfer an image to disk. By using this open source software, the documentation for preparing a disk image would be the same for Linux, Apple, and Windows. I've tested it with several distros on the pi so far and really enjoy the simplicity, even if I am comfortable using the `dd` command.

hocu, 2016-06-21:
I just installed Raspbian Jessie by using Etcher and decided to check if anybody's recommending it here. I used the Etcher only on Mac Os X 10.10 and 10.11.

It is pretty fast and easy.

bennuttall, 2016-06-21:
I'm fine with this. We've used it in a new starter guide that's going up soon: https://github.com/raspberrypilearning/help-software-guide/blob/master/quickstart.md

saildata, 2016-06-24:
I've only tested it from Arch and Debian hosts to Pi guests and full desktop ISOs; so far it's been great. @bennuttall, it looks like there isn't any other actions on this ticket before closing it, right? I'm assuming your code is the future [quick start guide](https://www.raspberrypi.org/help/quick-start-guide/)? 

Also, would it make sense to put a link pointing to the guide from the [downloads page](https://www.raspberrypi.org/downloads/), right at the first paragraph so that beginners can follow the logical flow of download -> get started? Something like  
> **Raspbian** is the Foundation’s official supported Operating System. Download it here, or use **NOOBS**, our easy installer for Raspbian and more. Once you download one of the distribution images, head over to the [Quick Start Guide](https://www.raspberrypi.org/help/quick-start-guide/) to get everything setup.

bennuttall, 2016-06-24:
Let's leave it open until at least a mention of Etcher has been added to the docs.

saildata, 2017-02-28:
@bennuttall  Going through my open/old issues - it looks like this is officially in the docs now [Download and image Raspbian directly step 8](https://www.raspberrypi.org/learning/software-guide/quickstart/)

Cheers --

lurch, 2017-02-28:
There's a difference between the info at https://www.raspberrypi.org/learning/ and the info at https://www.raspberrypi.org/documentation/ - this repo is about purely the latter.

saildata, 2017-02-28:
@lurch thanks for clarifying. I guess it's a little confusing since they're both on the org site.. I see now where there is some overlap and other parts, like this, that arguably should echo the same advice but perhaps don't.

JamesH65, 2017-03-06:
Etcher docs have been added for Mac, we are awaiting a PR for the windows version.

See #455, please use for future comments.

closed 2017-03-06 by JamesH65