raspberrypi/documentation #382: Zero mechanical drawing not up to date

ghent360, 2016-06-26:
Can you add the mechanical drawing for the Zero v1.3?

maurbeck, 2016-07-10:
I second this! v1.3 with the camera connector specs would be appreciated to help building cases that account for the camera cable, etc.

JamesH65, 2017-03-02:
I'll see what I can do to get this sorted. No timescales!

JamesH65, 2017-03-06:
@Roger-Thornton Also see #188, #95

JamesH65, 2017-03-09:
closing - please see linked issue #188

lurch, 2018-02-04:
Should this be re-opened? It was closed as a duplicate of #188 , and #642 claimed to close #188 , but #642 didn't add the PiZero 1.3 mechanical drawing that was being asked for here?

JamesH65, 2018-02-05:
Ah yes, we do plan on releasing it, but it won;t be right now - very busy.

JamesH65, 2018-07-02:
@Roger-Thornton This drawing too....!

JamesH65, 2018-09-12:
@Roger-Thornton Did we get anywhere with updating the mechanical drawings for the latest releases? https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/README.md seems to be missing the Zero 1.3, W and WH, Pi2 and the somewhat older but still sold A+.

JamesH65, 2019-02-07:
@Roger-Thornton Are there any plans to update/release any of these missing mech drawings?

JamesH65, 2019-02-07:
OK, just uploaded the Zero 1.3 files top github, should appear on website within an hour. We have no plans to add any for the Pi2 due to its age, so closing this ticket.

closed 2019-02-07 by JamesH65