raspberrypi/documentation #383: Configuring the Kernel - Update for cross-compilation

melvinisken, 2016-06-27:
I don't know if it is the right place to mention:
In the documentation of configuring the kernel (documentation/linux/kernel/configuring.md) it should be mentioned, that for cross compilation, the command 
`make bcmrpi_defconfig` has to be changed to
`make ARCH=arm bcmrpi_defconfig`
Otherwise it results in errors. It would be a nice hint for non-professional users.

lurch, 2016-06-27:
This is already covered on the 'building' page https://github.com/raspberrypi/documentation/blob/master/linux/kernel/building.md#cross-compiling - I dunno if it needs repeating on the 'configuring' page? \*shrug\*

MrAda, 2017-01-29:
I have tried both bcmrpi_defconfig and bcm2709_defconfig and neither seem to produce a zImage that works.  I have an RPI3.

pelwell, 2017-01-29:
Follow the instructions on the [building](https://www.raspberrypi.org/documentation/linux/kernel/building.md) page, using bcm2709_defconfig for Pi, and it should work. If not, report back with the actual error messages.

MrAda, 2017-01-29:
I am doing it again right now.  When I originally did the bcm2709_defconfig, the Pi would constantly reboot itself.

pelwell, 2017-01-29:
I would recommend updating your firmware as well - use rpi-update to get the latest kernel and firmware. You could also post your config.txt and cmdline.txt in case there are harmful or outdated settings.

MrAda, 2017-01-29:
I am doing this compilation from a VM running Ubuntu.  Will your "rpi-update" command even work?

pelwell, 2017-01-29:
It is the firmware on the Pi3 that might need updating - you haven't said what your current kernel is - so I was suggesting that you run rpi-update on the Pi to get the latest.

MrAda, 2017-01-29:
Well I downloaded a zip file.  Says linux-rpi-4.4.y.zip

MrAda, 2017-01-29:
Ok.  I removed the extra partitions and I am still getting a panic.  I'll reinstall Jessie's boot folder and overwrite with the new stuff to see if that works.

MrAda, 2017-01-29:
I am stuck.  Can't get past panic.  I took a picture with my phone to show what I am seeing.  Also I am getting some strange green lines too.
[image]
config.txt and cmdline.txt are standard jessie ones.

MrAda, 2017-01-30:
I completed reburned the jessie image and overlayed the new kernel on top.  That seems to have made it happy.  Still think it takes too long to boot up and I hope I can dig more into stream-lining the kernel.

JamesH65, 2017-03-02:
Is this covered by PR #465?

MrAda, 2017-03-03:
I don't know.  I have given up pretty much.

JamesH65, 2017-03-06:
#465 has been merged, and I think it fixes this issue so closing.

closed 2017-03-06 by JamesH65