raspberrypi/documentation #400: `dd` using Git-Bash on Windows - alternative to SourceForge/Win32DiskImager

ShadowLNC, 2016-07-18:
I noticed that Win32DiskImager is provided via SourceForge, and this concerns me based on the previous issues with SourceForge ([one article](https://notepad-plus-plus.org/news/notepad-plus-plus-leaves-sf.html) should suffice for those out of the loop).

I have successfully installed Raspbian to an SD card using `dd` (included) via Git-Bash (https://git-scm.com) on Windows, and the steps are almost identical (swap `df -h` out with `cat /proc/partitions`, and run Git-Bash as Administrator in lieu of `sudo`).

Given that Mac [already has three options for installing Raspbian](https://www.raspberrypi.org/documentation/installation/installing-images/mac.md), I figured this is quite similar, so I based my changes off the Mac page.
I forked the repository and made the changes, and then read `CONTRIBUTING.md` - so here's an issue and not a pull request (my commit also makes changes that are better suited to #185 and #350).

Side note: Windows 10's "Windows Subsystem for Linux (WSL)" (Canonical's Ubuntu binaries for Bash in Windows) don't currently have external drive access, and as such, `dd` through the WSL is not currently an option.

To summarise: Is it a good idea to add a (verified working) `dd` option to the Windows documentation?

XECDesign, 2016-07-24:
I am not sure if dd for windows is within the scope of the documentation.

> Note that this documentation is intended to be a short and concise set of helpful resources aimed at the majority of users. 

https://github.com/raspberrypi/documentation/blob/master/CONTRIBUTING.md

As far as the sourceforge concerns go, there's talk of using etcher (etcher.io) instead of win32diskimager.

ShadowLNC, 2017-02-14:
Having now had a chance to use Etcher, I would highly recommend it, but I do understand it's in beta. Given we're talking about the "majority of users", so not power users, would you suggest removing the OS-independent pages, and rewriting `/installation/installing-images/README.md` to include Etcher instructions? This would close 3 open issues in one fell swoop, and I'm happy to rewrite that documentation if it's deemed a good approach.

JamesH65, 2017-03-02:
Closing as docs are going to be updated to include Etcher.

closed 2017-03-02 by JamesH65

ITProggeruz, 2024-04-07:
just rufusify yourself with its dd method (rufus.ie)