raspberrypi/documentation #401: Pinout/hardware/interfacing information

arunkumar413, 2016-07-25:
Hi Team,

Over the years Raspberry Pi has evolved and various revisions have been made. However the documentation on them isn't properly maintained on the raspberry official website. I had discussed it on the forum (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=154936&p=1013261#p1013261
) regarding this. 

Could you please document the pinouts and hardware of different models on the official website. 

For example, I have a model B but I couldn't find a page which contains detailed information about the hardware/GPIO pinouts and how to interface external hardware with RPi.

Thanks,
Arun

waveform80, 2016-08-15:
The information as to the hardware available in each model is admittedly rather scattered over the official docs at the moment. However, the GPIO pin-outs are pretty well documented on the following two pages:

* [GPIO: Raspberry Pi Models A and B](https://www.raspberrypi.org/documentation/usage/gpio/README.md)
* [GPIO: Models A+, B+, Raspberry Pi 2 B and Raspberry Pi 3 B](https://www.raspberrypi.org/documentation/usage/gpio-plus-and-raspi2/README.md)

lurch, 2016-08-15:
#106 added a page just like you're asking for... and #132 & #182 enhanced it; but it later got removed again... https://github.com/raspberrypi/documentation/issues/182#issuecomment-83832608 and the following comments.

waveform80, 2016-08-15:
Hmmm ... that products page is nice (albeit difficult to find: as an experiment I tried locating it from the homepage; after exhausting several paths from the links at the top, I finally scrolled down :). Unfortunately it doesn't cover the historic models. Anyway ... might be nice to have a link to the products page from, perhaps, the hardware section of the docs?

lurch, 2016-08-15:
Ping @bennuttall as AFAIK he's responsible for the homepage navigation ;-)

bennuttall, 2016-09-30:
Yeah, not the most obvious but it's linked from the "shop" tag. And you'll find it in the homepage section on products, and in the footer.

lurch, 2016-09-30:
Doesn't seem to be in the footer any more?

bennuttall, 2016-09-30:
I stand corrected

Ruffio, 2017-01-07:
@arunkumar413 can this issue be closed?

JamesH65, 2017-03-02:
@bennuttall Anything further to comment on this? Can we close or is there more work to do?

JamesH65, 2017-03-09:
bump @bennuttall

bennuttall, 2017-04-04:
@JamesH65 Can we have a chat about this and other GPIO issues?

JamesH65, 2017-04-04:
@bennuttall Anytime.

JamesH65, 2017-05-16:
@bennuttall Still need to sort this out, got any time to discuss?

azrdev, 2017-11-12:
at least https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/ only documents Pi 1, without even mentioning later models (Pi 3 at least adds a second SPI interface, so there is a difference)

gnbl, 2017-11-17:
Also, "schematics" is not the appropriate term for what is available in the PDFs (e.g. https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/schematics/Raspberry-Pi-Zero-W-V1.1-Schematics.pdf) as these basically only show the connectors. And there is no explanation why these are so limited.

JamesH65, 2018-07-02:
@bennuttall Not sure where we are with this one. We could do with a decent pinout diagram in the docs - is there already one elsewhere on the site that's better than this one? https://www.raspberrypi.org/documentation/usage/gpio/images/gpio-numbers-pi2.png

azrdev, 2018-07-02:
@JamesH65 maybe just copy those from <pinout.xyz> (<https://github.com/gadgetoid/Pinout.xyz>). They are CC-BY-NC-SA, but maybe the authors are fine with official integration.

JamesH65, 2019-01-21:
@samalderhyde Ping re: incoming waffle ticket! This is the issues that has prompted the GPIO diagram request.

bennuttall, 2019-01-21:
The `pinout` command line tool shows the GPIO pinout of the Pi you're on, and can be run with a specified revision code to get others.

It might be worth just creating a markdown table of the pinout for each pinout, specifying which models the pinout covers. The source of `pinout` data is here: https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py

michaelstoops, 2021-03-18:
It looks like this has been addressed with `pinout` since https://github.com/raspberrypi/documentation/commit/3a76b6fd23fb77d4aee39b51185d023ed3991a8e.

Can we close this issue?

Cheers,
Michael

bennuttall, 2021-03-18:
I'd say so 👍

JamesH65, 2021-03-18:
Closing this issue as questions answered/issue resolved.

closed 2021-03-18 by JamesH65