raspberrypi/documentation #412: raspiyuv: padding 16 -> padding 32; see issue raspberrypi/userland#329

reiner-dolp, 2016-08-12:
According to the current docs `raspistillyuv` is padding images to be divisible by 16. However, it actually pads the image to be a multiple of 32. See issue raspberrypi/userland#329 for details.

I already opened pull request #411 with the trivial fix.

lurch, 2016-08-12:
No need for both a PR and an issue, the PR alone is enough :)

closed 2016-08-12 by lurch

Ruffio, 2016-08-12:
@lurch if an issue is not required, then perhaps the frontpage (https://github.com/raspberrypi/documentation) should be changed, as it says:

>  If you have anything to fix or details to add, first file an issue on GitHub to see if it is likely to be accepted, then file a pull request with your change (one PR per issue).

reiner-dolp, 2016-08-12:
@ruffio @lurch also: https://github.com/raspberrypi/documentation/blob/master/CONTRIBUTING.md

lurch, 2016-08-12:
`CONTRIBUTING.md` also says:

> With the exception of typos and spelling mistakes...

While it's not a typo, I believe a small technical fix like #411 implemented kinda falls under the same category (especially as it was already confirmed by @6by9 in the userland repo).