raspberrypi/documentation #414: Add details for access-over-internet service

timfernando, 2016-08-15:
My first contribution here, I'm proposing an addition/deletion to [access-over-internet](https://github.com/raspberrypi/documentation/blob/master/remote-access/access-over-Internet/internetaccess.md) to better explain the already-listed remote access service Dataplicity and how it works. 

I'm adding an issue here as per contribution instructions, but as the change is minor I've also the PR for it. Hope that's OK 👍

lurch, 2016-08-15:
There's ongoing discussion on the relevant PR, so closing this issue.

closed 2016-08-15 by lurch

timfernando, 2016-08-15:
I believe the discussion is complete as the requested change has been made. Should the merge request go in the PR or in the issue?

lurch, 2016-08-16:
The PR (Pull Request) *is* the merge request ;)