raspberrypi/documentation #423: documentation for cross compiling broken..

JoanTheSpark, 2016-09-06:
"You can then copy the following directory to a common location /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian, and add /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin to your $PATH in the .bashrc in your home directory."

This sentence doesn't make sense and only after an hour of googling I found an older post on the RPforum that had two more words in there:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=124061
"You can then copy the toolchain to a common location **such as** /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian, and add /tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin to your $PATH in the .bashrc in your home directory."

Now, that makes sense..
Please put them back in.

JoanTheSpark, 2016-09-06:
And while you're at it.. it's not enough to just add `/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin` to the PATH variable.
In my case Ubuntu 16.04LTS 64bit in Virtual box it needed this path to actually work:
`**/home/<my user name>**/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin`

lurch, 2016-09-06:
See also #407

JamesH65, 2017-03-02:
Closing - latest docs have been updated (or will be very soon). Any further issues, please reopen (but not right away please to ensure updates make it through the system)

closed 2017-03-02 by JamesH65