raspberrypi/documentation #424: Memory Split on gpu-mem under 32GB influences MPG2 license activation output of vcgencmd

FlyinFistOfJuda, 2016-09-11:
Hello,

just lost some grey hair today & a few bucks on the raspi-foundation, but gained some wisdom:

I had my GPU Memory Sharing (gpu-mem) set to 16MB when i bought a MPG2 License.
Registration went fine, but the output of vcgencmd codec_enabled MPG2 kept returning DISABLED.

THE gpu-mem value has to be a minimum of 32GB for successful activation. Would be nice if this gets a place in documentation and/or additionaly in the communication from sales in the installation guide.

Cheers!

lurch, 2016-09-12:
This was already partially addressed in #117 ?

FlyinFistOfJuda, 2016-09-12:
That would explain my out of memory errors. Shouldnt open bugs that late. :)
I've read #117 , it fits, but perhaps you may consider making this "problem" more obvious to the users who try to enable mpg2. Anyhow, guess they will find this bug report. For now, i'd leave the decision to you guys and close this one.

Thanks & Reg.

closed 2016-09-12 by FlyinFistOfJuda