raspberrypi/documentation #434: BOOT FROM A USB MASS STORAGE  need a little adjust

congzhangzh, 2016-10-01:
the document https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md need some minor fix.

As sudo sed -i "s,/dev/mmcblk0p,/dev/sda," /mnt/target/etc/fstab was not enough, assume the old part ion as /dev/mmcblk0p1 and /dev/mmcblk0p2 is not right,  for example, on my sd card, it's  /dev/mmcblk0p6 and /dev/mmcblk0p7(this cause by OOBS?) the user may need a check!

hoall, 2016-11-15:
In addition, is there still the need to use the branch next or is it already merged into master?

bstabens, 2017-01-04:
May I suggest to add a description to the document mentioned above how to boot from usb stick while having multiple usb disks present?
Like, adding the root=PARTUUID=%partitionuid% to /boot/cmdline.txt and using /dev/disk/by-uuid/%filesystemuuid% in /etc/fstab?
As it may not be uncommon to have multiple usb mass storage devices running at a pi.

Thank you.

sschadwick, 2017-01-06:
@congzhangzh What command did you use to fix the issue?

IgorGanapolsky, 2017-01-14:
I followed instructions here: https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

However, the Pi3 just shows a blank blue screen.  It does not boot up at all.  I think your instructions are wrong, as the SD card needs to be inserted in the Pi at all times...

Please advise,
Igor

sschadwick, 2017-01-15:
@IgorGanapolsky Did you get any error messages during the install?

I had problems getting Raspbian to boot, it would boot up to the raspberry logos but hang. That issue somehow resolved itself overnight, I didn't change any settings. The pi now boots up into a desktop, but wlan0 is not recognized so I can't connect it to wifi.

IgorGanapolsky, 2017-01-15:
No error message, just a blue screen.  That happens when there is no SD
card present

mopenmind, 2017-02-03:
It would be great to know if branch next is still needed or is it already merged into master.

lyq1996, 2017-02-12:
I had problems boot from usb disk too.
It still need SD card.If I have't insert SD card,it cant boot.

congzhangzh, 2017-02-19:
What I got from my test:
1. USB will be too hot, which seems not be what it should be, so I stop plug a USB in my pi. If you has no trouble, you can go further more!
2. when you want boot from USB failed which follow the official guide, first check that /mnt/target/etc/fstab on the USB drive,  it should be easy to fix, just make the fstab fit your USB disk layout
3. if you fstab is right, but pi does not boot from USB, you may try to switch to another USB port(you may get different result, I think the pi 3 has too much usb device like ethernet、bt、wifi、etc. which make boot step a litter more hard!) , good luck
4. if you failed even, you may boot from SD card, and check you USB is dejected or not. if your raspbian does not found it (and you are sure you disk is OK), don't worry, find a Linux PC, plug the USB disk, the PC should found it and some internal USB state(USB disk is controller + disk, right?) may be reset, plug back to you pi and now it should work(or you can use the fixed USB disk to boot you pi from USB disk, change port if needed, the prepare process maybe need once again if your pi reboot), good luck.

Tks.

JamesH65, 2017-03-08:
You no longer need to do an rpi-update, the standard firmware has been updated to include the required options.

@ghollingworth Any other comments?

fourthhorsem4n, 2017-03-16:
I can't get this to work at all. I've tried with two different external hard drives and two different thumb drives. Followed the instructions exactly as shown.

JamesH65, 2017-03-16:
Did you try any of the tricks from the comments above, or just the standard docs?

fourthhorsem4n, 2017-03-16:
I did try all the tricks from the comments here as well and still can't get it boot from usb. Just get a blue screen as mentioned by others.

fourthhorsem4n, 2017-03-16:
I was mulling this over and it's just a thought but maybe I need to increase usb time out in the boot configuration on the usb drive?

achurak, 2017-03-19:
Is there a way to increase the boot time to something even more than 5 seconds? I've a feeling that my 2TB WD My Passport is just not able to start spinning fast enough for the pi to boot from it hence I see just the black screen no matter what USB port I try (and I do I use externally powered USB-hub). I know I've followed the guide correctly because I'm able to boot from a 32GB thumb drive (though I do have to use the boot file) which I configured in exactly the same way as the HDD.

JamesH65, 2017-03-20:
@ghollingworth Think you are the best one to look at this.

ghollingworth, 2017-03-20:
Have you tried booting from an SD card with just bootcode.bin on it?  To enable the extended delay also create a file TIMEOUT doesn't need to contain anything just the existence of the file is enough.

Gordon

fourthhorsem4n, 2017-03-20:
I have not tried that yet. One thing I did notice in the documentation is when creating the new fstab it says to use /dev/sda however I'm not sure if this is entirely correct because we now have two file systems on the USB thumb drive respectively /dev/sda1 and /dev/sda2

achurak, 2017-03-20:
Can anybody please confirm whether WD or any 2TB HDDs are supposed to work with this configuration. I tried everything, re-did the whole setup process 3 or 4 times already and it still won't work - show just black screen even though I do feel the HDD spinning.

fourthhorsem4n, 2017-03-24:
I was able to get this working on a thumb drive but not an external hard drive by placing bootcode.bin on a fat32 formatted microsd card

achurak, 2017-03-24:
Same here, works perfectly with a thumb drive, but not hdd. Surprisingly I can boot from hdd when the thumb drive is inserted. It looks like it needs it to even start booting and then it can boot from anything. Another weird thing is that it doesn't boot from any usb device as long as my powered usb hub is connected to it, doesn't even matter if I have the boot device connected to the local usb port, plus anything connected to the hub wouldn't even start (judging from the led lights on the thumb drive). If I disconnect the hub either from the raspberry or from the power, power cycle the raspberry, wait for the rainbow on the screen and plug the hub back in, it starts to work perfectly, now raspberry can even boot from devices connected to this hub. All in all I find this behavior very strange for a feature that's been released more than half a year ago.

sschadwick, 2017-03-24:
Can you confirm that the HDD is receiving enough power? Does the drive have a dedicated power plug or does it draw from USB?

I haven't attempted to boot from 2.5" HDD, but I can confirm there are significant issues mounting a drive if your raspi isn't getting ~2 amp from USB. The way I understand it, both the pi and drive require about an amp of power each.

fourthhorsem4n, 2017-03-24:
Both of my external hard drives are powered via USB and run on a powered USB hub

JeffReeves, 2017-03-25:
I'm getting the same issue as this person:

https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/#comment-1267889

--- 

Basically I follow all of the steps in the official guide, and when I get to unmounting the /dev directory (`sudo umount dev`) it gives this error:

`Umount: /mnt/target/dev: target is busy`

I did a lazy unmount with the -l flag on the umount command to get past the error, and completed the remaining steps. When I go to boot from USB I get through the rainbow screen just fine, but then it gets stuck at the screen with the 4 raspberries. I can hit Enter and see a blinking _, hitting Ctrl+C shows the ^C but doesn't do anything. So the screen is responsive and not frozen, but it doesn't go anywhere.

fourthhorsem4n, 2017-03-25:
@JeffReeves as for the unmounting issues make sure the USB thumb drive is unmounted prior to starting this process as I have ran into the same issue before and it was because I had forgot to unmount the USB thumb drive.

JamesH65, 2017-04-20:
I believe that the new msd docs now should cover the point above, so am going to close this. If there are still issues, please reopen this, or perhaps better to start a new issue with specific description of the problem.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

closed 2017-04-20 by JamesH65