raspberrypi/documentation #437: Raspberry Pi Schematics -- USB

noafterglow, 2016-10-05:
I hate to do this but I'm asking for more schematic information on the Pi 3.  Reason is I'm planning to embed them in a product, and the power system for the usb ports is an issue.  I need questions answered like:

1) are the ports protected against overload?  
2) are they individually protected or ganged?
3) are the parts used "UL (insert some random request from UL)"
4) Etc...

I can't do that without more schematic and parts list info.

Thanks

VaclavSal, 2016-11-08:
My first post so if this is  not proper place let me know. .
I am just looking into switching to this product and have same question. 
How are USB ports powered / connected ?
Also did not see SPI  or I2W in latest  3B version. 
Not that it would be that important , but is hardware "Open source ?  Publishing limited schematic is not a good indicator about "Open source "

lurch, 2016-11-09:
> Also did not see SPI or I2W in latest 3B version.

SPI, I2C, 1-Wire, etc. are all enable-able via the options in the Raspberry Pi Configuration dialog ( https://github.com/raspberrypi/rc_gui ), which is included by default in Raspbian. See e.g. the screenshots in https://www.raspberrypi.org/blog/another-update-raspbian/

It's been mentioned elsewhere that the hardware is **not** Open Source (and therefore I have no idea about your USB question, other than to say that the single USB port on the Pi's SoC (BCM2837 on Pi3B) is connected to a LAN9514 chip which provides an extra 4 USB ports and an Ethernet interface).

JulyJim, 2016-11-09:
Thanks for reply.Please advise where is the proper company sponsored place to ask similar questions. 
I am speaking from bad  experience conversing on forums with well meaning users / volunteers where wild guess / misinformations were common. 
In this case it seems that most info on USB is on single port Pi which does not help much to make conciseness decision on purchase. 

PS 
I would also appreciate in future to NOT to have my e-mail  CC:Please use BCC if you have to. 

   

SPI, I2C, 1-Wire, etc. are all enable-able via the options in the Raspberry Pi Configuration dialog ( https://github.com/raspberrypi/rc_gui ), which is included by default in Raspbian. See e.g. the screenshots in https://www.raspberrypi.org/blog/another-update-raspbian/It's been mentioned elsewhere that the hardware is not Open Source (and therefore I have no idea about your USB question, other than to say that the single USB port on the Pi's SoC (BCM2837 on Pi3B) is connected to a LAN9514 chip which provides an extra 4 USB ports and an Ethernet interface).—

bennuttall, 2016-11-09:
> I would also appreciate in future to NOT to have my e-mail  CC:Please use BCC if you have to.

Nobody sent that email. It's automatically sent from github when someone comments in reply to you. Nobody else sees your email address in the CC. But you just posted it publicly on the github issue.

JamesH65, 2016-11-09:
(Quite ironic really)

The 3B has all the same interfaces as the 2B.

You are likely going to get more responses on the Raspberry Pi forum than
here, the forum has many more viewers.

When it comes down to it, the Pi3B costs $35. You are spending more than
that in time asking the questions that might be more easily answeed just
getting one and trying out whatever you want to do.

JulyJim, 2016-11-09:
OK, no problem

JulyJim, 2016-11-09:
Pretty bad start.
Still not sure if this is right place to ask questions, but getting the feel it is not. 
Apart of "it cost $35 " I realize that  people  feel obligated to chime-in with stuff they feel is relevant. 
No, I do not need any of such advises and if I do I will say so. 
I'll stand by for somebody to get real and answer the OP I have tagged on. Sorry - did not mean to hijack the discussion with my stuff. ( Yes, I'll check the "users forum ". )

JamesH65, 2016-11-09:
As I said above, the Raspberry Pi forum would be a better place to ask questions. The schematics are not available, this is not Open Source hardware. Unlikely to get any more that that there.

If building in to a product, you could use the Compute module, schematics are available for that I believe, along with integration details. 

I'll close this now since there is nothing else to say.

closed 2016-11-09 by JamesH65