raspberrypi/documentation #444: How to set program_gpio_bootmode and program_gpio_bootpos not documented in bootmodes.md (or anywhere else)

datajerk, 2016-10-19:
Subsequently, the boot ROM checks to see if program_gpio_bootmode OTP bit is set, if it is then it reads either GPIOs 22-26 or 39-43 (depending on the value of program_gpio_bootpos) and uses those bits to disable boot modes. 

Docs not clear how to set the above.

nos86, 2016-11-16:
Hello guys,
I have the same problem. Someone, can help us to under how to control boot sequence via gpio?

Thanks in advance,
Salvo

rbnpi, 2016-11-17:
Yes please
More info on the GPIO pins to use to disable SD card boot and thus remove the 5 sec delay and go straight to USB boot. I don't want to have to put a blank SD card in instead

TheDiveO, 2017-03-07:
Dear Raspberry Pi Foundation,

while booting from microSD is first, and booting USB mass storage devices is second, can we please make quick booting from USB MSD great? You see, booting from mSD. So awful!

Any contemporary jokes aside, please finally document `program_gpio_bootmode` and how to use the GPIO pins in order to skip waiting for the microSD boot attempt to time out.

JamesH65, 2017-03-08:
@ghollingworth Any thoughts on this?

ghollingworth, 2017-03-08:
OK, yes I've implemented the functionality but haven't properly tested it, will have a look at it tomorrow to see if I can get it working...

wezhunter, 2017-03-24:
@ghollingworth I'd be very very interested & really appreciate having the gpio_bootmode docs updated to reflect actual usage for a potential medium-large scale CM3 project I'm working on. E.g SPI primary boot mode which I've heard you mention that works (SPI flash u-boot bootloader)... Many thanks

TheDiveO, 2017-04-07:
@ghollingworth any updates for the documentation, and `program_gpio_bootmode` in particular?

andrum99, 2017-04-08:
+1. I'm looking to boot from USB without the 5 second delay. It would be useful to know how I can do this by programming the OTP.

JamesH65, 2017-04-20:
@ghollingworth Anything further on this yet?

chrisR88, 2017-05-05:
I believe I have found another worry in not understanding the setting of the GPIO boot_pos. In my current design using the RPi, I am using gpio 22- 26.  Some of these pins are pulled low but some are high on power up. This means that voltages are present before the ROM starts to boot. The confusion that I have on this subject is giving me concerns about my pcb design. The low tech solution is to remove all the high pins to another gpio pins out of the range 22-26 by retracking the pcb.   My plan is  not to use the sd card, but  a ssd but as has been previous mentioned  it the fault recovery process  that is important. I would like to be able to revert to the  sd card in the case of failure to boot from the ssd.  To do this I was planning to use one of the switches which is normally low , so that pressing the switch and powering up the system would force a sd card boot.  But which GPIO pin do I use?  The speed of the boot up is important, so the  best recommendation I have  seen  is a blank sd.

JamesH65, 2017-05-16:
ping @ghollingworth

JamesH65, 2017-06-06:
ping @ghollingworth. Is more documentation required here?

ghollingworth, 2017-06-06:
Yes there is, but I've not recently had the time to spend documenting it...  I may be able to get around to it soon

ghollingworth, 2017-06-29:
Fixed documentation to show that actually there is only one config parameter for this.

Add the following to config.txt to enable GPIO bootmode from 22-26 (program_gpio_bootmode=1) or 39-43 (program_gpio_bootmode=2)

```
# Enable GPIO bootmode from GPIOs 22-26
program_gpio_bootmode=1
```

Since  this is programmed into the OTP it'll take effect (and cannot be undone) after a subsequent power cycle.

Gordon

closed 2017-06-29 by ghollingworth