raspberrypi/documentation #447: Tell Users the Name of Desktop GUI

PeteCress, 2016-10-29:
I have been through all the topics on https://www.raspberrypi.org/help/, but cannot find anything that tells me how to find out which Desktop GUI I have on my Pi.  

Most recent guess was LXDE, but right-clicking the desktop does not offer up an option to create a shortcut... so I guess it's some other GUI.

If there is no place in Raspbian that will tell me, maybe one of the Sofltware Guide screens could have a grid that shows which GUIs are supplied with which versions of Raspbian.

Oh yeah...tangentially.... I have also been unable to find anythng in Raspbian that tells me the version number of the system.   Probably there and I'm just not seeing it... but nothing jumps out at me like a _Menu | About Raspbian_ option on the desktop.

JamesH65, 2016-10-30:
It's always been LXDE, slightly modified. Latest version is more modified and called Pixel. Still LXDE underneath. 

Typing in uname -a in the terminal will return a few versions numbers.

lurch, 2016-10-30:
I think https://www.raspberrypi.org/blog/introducing-pixel/ answers a few of your questions :-)

closed 2016-10-31 by bennuttall