raspberrypi/documentation #450: Add link to Etcher and also more explanation for unzipping using eg 7Zip for example

pageauc, 2016-11-02:
Reference page https://github.com/raspberrypi/documentation/blob/master/installation/installing-images/windows.md
The windows PC procedure you outline assumes SD card does not have a previous OS image burned on it.  Quite often the SD card will have another image that the user is trying to overwrite or replace.  In that case there is only a small fat32 partition eg Raspbian.  The user would have to understand this and know how to delete the unix partitions using gparted or some other tool and reformat SD card to fat32.  This can take some time. Win32DiskImager does not repartition the SD card.  In the past I have used my Cannon Camera  to clear all partitions and reformat complete SD card back to fat32..  An alternative windows utility that is easier to use is Etcher https://etcher.io/  This utility will seamlessly clear all partitions and install the new image with little input required from the user.  This option should be added to the documentation.
Also it is all fine a good to say unzip a file but I believe a little bit more explanation is required or a link to alternative step by step procedure added.  I have found the best tool to be 7Zip.
The procedure does assume some knowledge by the user but there needs to be either links to good tutorials for less knowledgeable users or include a separate doc page. IMO

lurch, 2016-11-02:
There's also https://www.raspberrypi.org/learning/software-guide/
I've never been quite sure how the various learning guides seen on https://www.raspberrypi.org/help/ are supposed to supplement / enhance / replace the instructions in the [documentation](https://www.raspberrypi.org/documentation/) ?

pageauc, 2016-11-02:
Well the software-guide per link is for Raspberry Pi applications that .  There is no mention of initial image burning or associated software but under link https://www.raspberrypi.org/learning/software-guide/quickstart/ there  is mention of etcher.  This link would be very suitable as a reference in the github doc page

Ruffio, 2017-01-07:
Xref.: https://github.com/raspberrypi/documentation/issues/455

JamesH65, 2017-03-02:
Closing - we already have a issue for Etcher and it will be put in the docs.  #455

closed 2017-03-02 by JamesH65