raspberrypi/documentation #451: Raspberry Pi 3 power layout

christopherlange1110, 2016-11-06:
We need a layout for the usb power please. If not can you tell me the value of r31 at least so i can fix my board?

christopherlange1110, 2016-11-06:
I know this hardware is closed source even a full/partial schematic with a few key componets blurred out to keep the cloners at bay. even that would help us greatly.

lurch, 2016-11-06:
If you've "broken" your board somehow (?) can you be sure that simply replacing r31 will be enough to fix it?
I assume you've already seen https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md ?

christopherlange1110, 2016-11-06:
Well it was working perfectly and I have received one of them layer cases for it and put it in the case it worked until I moved it to another location. When I powered up the board I noticed the issue.  So I removed it from the case on a antistatic mat I noticed the resistor fall out I then proceeded to examine the board with a magnifying glass to see if there was more damage. Not seeing any other damage I was satisfied with that replacing the resistor would fix the issue.

Doug-Wyman, 2016-11-07:
If the pcb is cracked and is I presume a multi layer board, the odds are not in favor of the resistor being the problem.

lurch, 2016-11-07:
> one of them layer cases

Is it possible you used a case designed for the Raspberry Pi 2 with the Raspberry Pi 3? Most cases should work fine with both, but if it's _particularly_ tight-fitting, it might only work with one or the other?

christopherlange1110, 2016-11-07:
Can I please just have the value of the resistor so I can at least try at failing?  I tried to get info in the forum and was sidelined. This was I thought the best place to look

pelwell, 2016-11-07:
Any multimeter should be able to measure the resistor, or you could just solder it back, but you won't get any official information except that which has already been published.

christopherlange1110, 2016-11-07:
That stinks. Kinda sucks to to throw away a board for a .001 cent part. It's has disappeared from my mat and vanished on the floor? Can you guys at least tell me if this resistor will cause the USB Wi-Fi and card act led and Ethernet to stop working? It still boots and runs but you joy on connecting to the outside world.

pelwell, 2016-11-07:
For what it's worth, I unplugged my Pi3 and stuck a multimeter across R31 - it said 0 Ohms.

JamesH65, 2016-11-07:
Was looking for R31 on the board so I could measure it, as per pelwell above, but cannot find it anywhere! However, did note that R42 is missing on my PI3. Doesn't seem to make much difference.

Note, you are not throwing away a board because of a 0.001 cent part, you are throwing it away because of bad handling practice. This sort of thing happens all the time. Throw away a $500 car, because of a $1 part. That costs $500 to fit.....

pelwell, 2016-11-07:
It's very close to a shiny package on the underside of the board.

JamesH65, 2016-11-07:
Thanks Phil, I concur with 0 Ohms.

JamesH65, 2017-03-02:
Not the place for tech support....

closed 2017-03-02 by JamesH65