raspberrypi/documentation #455: Include instructions for Etcher

johannrichard, 2016-11-19:
[Etcher](https://etcher.io) is a graphical, cross-platform tool that is by far the easiest way to flash disk images to SD cards (also for Mac OS X which has been short of suitable, graphical installers so far).

It would probably be helpful to have a short instruction on using Etcher in the documentation.

[image]

Ruffio, 2017-01-07:
@lurch what is the opinion of introducing Etcher here?

brianjking, 2017-02-17:
@Ruffio -- I highly suggest it! It makes it very easy to add the `ssh` folder to the `boot` partition to enable headless connection via `ssh` to the Pi with the recent changes to the Raspbian distribution releases.

I opened a PR #511 for this on the MacOS page and as I mentioned in the pull request if this is merged in I'm happy to write instructions for Windows as well as this is also proposed in #450.

Thanks!

Ruffio, 2017-02-17:
@brianjking I'm not within the foundation and have no ruling :-( You must convince them in order to get your RP accepted.

brianjking, 2017-02-17:
@Ruffio Sorry - updated post & tagged @lurch

JamesH65, 2017-03-06:
The MAC docs for Etcher have been merged - would it be possible for a Windows version to be written as suggested above? @johannrichard

JamesH65, 2017-03-06:
Actually, Etcher on Linux should also be documented if you have the time.

lurch, 2017-03-06:
Only needing one set of instructions for all three platforms, was part of the original reason behind Etcher ;-)
https://resin.io/blog/introducing-etcher/

JamesH65, 2017-03-07:
I wonder if we should have one page for Etcher, and link to it from the other OS specific pages?

JamesH65, 2017-03-07:
After thinking about it for another 2 minutes, Yes, I think that is the way to go - one Etcher page, then link to it from the individual OS installation pages. So if @brianjking fancies knocking up a PR to do that, that would be splendid.

pimterry, 2017-04-11:
Looks like this has gone quiet - anybody object if I pick this up?

JamesH65, 2017-04-11:
Feel free!

pimterry, 2017-04-11:
@lurch @JamesH65 This is now done, see #614.

JamesH65, 2017-04-19:
Fixed by #614

closed 2017-04-19 by JamesH65