raspberrypi/documentation #464: Display page is messed up at the bottom

mrichardson23, 2016-11-30:
https://github.com/raspberrypi/documentation/blob/master/hardware/display/README.md

At the bottom of the page, beneath the headline "MODULE MECHANICAL SPECIFICATION" seems a bit weird. The list stops and there's a non-sentence and what possibly should be another list item?

lurch, 2016-12-01:
https://github.com/raspberrypi/documentation/commit/f007cadf3536cda8b87ce4649feb6d51a86bf189

ping @ghollingworth

LornaLynch, 2016-12-01:
I looked at it a while ago but assumed it wasn't finished yet.

ghollingworth, 2016-12-01:
Edited again...

LizUpton, 2016-12-01:
Opening again: @ghollingworth, please don't make edits without putting them past the copyeditors! @LornaLynch, could you please do the honours?

Ruffio, 2017-01-07:
@mrichardson23 this can be closed, right?

JamesH65, 2017-03-02:
looks OK to me. Closing.

closed 2017-03-02 by JamesH65

mrichardson23, 2017-03-02:
Sorry! Missed this mention. Thanks all.