raspberrypi/documentation #466: Trouble enabling VNC Server at the command line

JordanLyons, 2016-12-10:
I am following the [documentation](https://github.com/raspberrypi/documentation/blob/master/remote-access/vnc/README.md) and "Advanced Options" in my raspi-config does not include a VNC option. I am using Jessie/Pixel/NOOBS.

I'm very new to Raspberry Pi, so I'm not sure if this is just my issue or something worth addressing in the documentation.

waveform80, 2016-12-22:
Have a look under "Interfacing Options" instead of "Advanced Options". I think there's another ticket somewhere correcting that page but it evidently hasn't been merged yet...

waveform80, 2016-12-22:
Ah, my mistake - #472 corrected the documentation for SSH, but not VNC. I'll bung a quick PR together to fix that too.

waveform80, 2016-12-22:
Opened #476 to fix the VNC docs

LornaLynch, 2017-01-03:
Thanks: I've merged the PR.

closed 2017-01-03 by LornaLynch