raspberrypi/documentation #471: Why disable SSH by default?  Makes life difficult for disabled people.

escapologyBB, 2016-12-14:
Hi guys,

Just a quick question about why you disabled ssh by default on all new installations of raspbian?  I own more Raspberry Pis then is probably healthy for one man and have spent all morning trying to get to the bottom of this problem, you see an quadriplegic and just swapping and SD card when there's a problem involves my asking an able-bodied person for help which isn't really that independent, is it.  You need to make it much more obvious about this little file we have to put on the boot partition of the SD card, because I can't just go and plug-in a keyboard and monitor and diagnose the problem that way.

I useRaspberry Pi's all over my house for home automation amongst other things, so without easy access to ssh I'm absolutely knackered.  I'll happily help with the documentation but you really will disenfranchise a lot of disabled people if you don't make this change more of yes, or ginger back of course but I don't think that's likely.

Can somebody go into detail as to why this change was made these?

Many thanks!

JamesH65, 2016-12-14:
Having SSH open by default has proven to be a big security issue. Please see the blog posting on this.

https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/

JamesH65, 2016-12-14:
Also worth noting, you can add a simple (empty) file to the /boot folder to enable SSH automagically. See blog again.

closed 2016-12-14 by bennuttall

jsavage, 2016-12-15:
Stuart,
I have a number of headless pis too so my initial reaction was similar.
However,  this update is not retrospective so existing pis are not affected
as far as I can see and ssh can be enabled before the sd card is first
inserted in a pi.  Do make sure you change the default password though.
JS

bennuttall, 2016-12-15:
This was explained in the [blog post](https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/). This is not the place to discuss the change. Please take discussion to the [forums](https://www.raspberrypi.org/forums/).