raspberrypi/documentation #474: Documentation for Boot Process from SD Cards

machinaut, 2016-12-16:
I'm trying to figure out a good strategy for using high capacity (64GB+) SD cards and right now I'm stuck doing trial-and-error with reformatting and reflashing and booting.

Could we get some documentation on what it reads where and how?

lurch, 2016-12-16:
Does this help? https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

Or perhaps you're looking for more low-level details (although I'm not sure how that'd help), in which case you may find this useful? https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained

And of course there's also https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/README.md

JamesH65, 2017-03-02:
@machinaut I'll be closing this issue unless there is still an issue. Did the links above solve your problem?

JamesH65, 2017-03-08:
No response, closing.

closed 2017-03-08 by JamesH65