raspberrypi/documentation #486: Raspbian Lite vs Full Raspbian

earlthesquirrel, 2017-01-16:
I've tried 6 times with the latest OS releases and just FYI the 5 times with the full release I get an error during the rsync (which seems to unmount the partitions).  With the Lite version it worked 100% as advertised. 

I think there might be something that's changed in the full release that's causing an issue in copying the info..

lurch, 2017-01-16:
I think you need to provide more context about what it is you're referring to...

earlthesquirrel, 2017-01-16:
When following the instructions for How to boot from a USB Mass Storage Device on a Raspberry Pi 3, starting with an empty SD card and empty SSD (or USB drive), I encounter an issue when using Jan-11-2017 release of the full Raspbian install.

I can get thru the instructions for updating the firmware and the pi reboots and I can validate everything is good.

When I attempt to do this : sudo rsync -ax --progress / /boot /mnt/target

I get an RSYNC permission error, saying that partition is RO.  When I check, at least one of the /mnt/target partitions is unmounted.   I've tried with different USB and SSD.  Occurs every time.

I finally after 5+ failures, tried to follow the same instructions with the only change being that I started with Raspbian Lite.   The instructions worked flawlessly.

Given these instructions were initially constructed for an earlier release, I'm thinking something has changed that causes the rsync to fail.

lurch, 2017-01-16:
Ping @XECDesign and @ghollingworth

ghollingworth, 2017-01-16:
When you get to this stage what do you get from lsblk?

JamesH65, 2017-03-02:
This not a documentation issue. If the originator still has the problems, please use the forums where help with be forthcoming.

closed 2017-03-02 by JamesH65