raspberrypi/documentation #490: Using `wpa_passphrase` for generating secret token instead of entering password as plain text

aleksandaratanasov, 2017-01-20:
Many tutorials (including the official documentation here) store the password for a profile inside `wpa_supplicant.conf` as plain text. This is wrong since it not only doesn't mention the security concerns related to this but also doesn't nurture a more security-oriented approach from the user when handling such information. In addition `wpa_passphrase` ensures a minimal length for a password (8..63 characters) which adds to the security enhancement it already offers.

Check my edit off this article [https://github.com/aleksandaratanasov/documentation/blob/master/configuration/wireless/wireless-cli.md](url).

lurch, 2017-01-20:
Saying "check my fork" isn't very useful - if you have changes to suggest, please submit a pull request.

aleksandaratanasov, 2017-01-20:
The guidelines state that an issue has to be created first and then a pull request. Believe me, I wouldn't have bothered posting this otherwise. :) I'll create a pull request now.

commit dc8c0400 (2017-03-02): Merge pull request #491 from aleksandaratanasov/master

Fixes #490 by adding small introduction to using wpa_passphrase with wpa_supplicant configuration
closed 2017-03-02 by LornaLynch