raspberrypi/documentation #495: config-txt voltage info needs updated

portets, 2017-01-22:
Specifically sections relating to the over_voltage and force_turbo flags. The Pi 2 and Pi 3 aren't 1.2V, they are 1.3125V. This means the max over_voltage is 4(1.4V) instead of 8(1.4V), with or without force_turbo=1. Many overclockers are using values >4 to no benefit.

Also current_limit_override is enabled by default now from what I've read? If so, then it should be removed from the documentation.

JamesH65, 2017-03-06:
@pelwell I am not familiar with this stuff - is this correct?

JamesH65, 2017-03-09:
@pelwell bump - or do I need to talk to @popcornmix?

pelwell, 2017-03-09:
@popcornmix is the expert, but from what I can tell Pi2 and Pi3 start at 1.2 + 18*0.00625 = 1.3125V and Pi Zero (inc. W) start at 1.35V.

However, over_voltage is in units of 25mV relative to the common base of 1.2V, so over_voltage of 8 is always 1.4V, but Pi 3 runs with an effective minimum over_voltage of 4.5 (6 for Pi Zero).

pelwell, 2017-03-09:
N.B. over_voltage is only applied when turbo mode is active, otherwise the base voltage is 1.2V across the board.

JamesH65, 2017-03-09:
@pelwell Phil, would you mind giving the #569 patch a once over, specifically the added voltage table, to ensure I understand your comment above.

pelwell, 2017-03-09:
It looks OK to me, but wait for @popcornmix's approval before merging.