raspberrypi/documentation #498: Compute Module display page needs dt-blobs updating for CM3

6by9, 2017-01-23:
The dt-blob files associated with https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/cmio-display.md need to be updated for CM3. Currently they only work with CM1. 

The blobs on the camera page got updated, but not the display. 
https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=172421&p=1103246 has confirmed that simply adding
```
   pin_defines {
       pin_define@DISPLAY_SDA {
          type = "internal";
          number = <0>;
       };
       pin_define@DISPLAY_SCL {
          type = "internal";
          number = <1>;
       };
       pin_define@DISPLAY_I2C_PORT {
          type="internal";
          number=<0>;
       };
```
appears sufficient. Probably ought to do a slightly more comprehensive test though.

6by9, 2017-02-02:
In updating the blobs I see that the current configuration differs from the standard wiring for the camera modules - the camera wirings have been swapped, and it uses GPIOs 4&5 instead of 2&3.

In the interests of consistency and being able to swap directly to the blobs from https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/cmio-camera.md, I'm going to update to the "standard" wiring, but that also means a new set of pictures to be taken.

JamesH65, 2017-03-06:
@6by9 Is there anything that needs to be done to close this issue?

JamesH65, 2017-03-06:
Ah, think I see what  you are getting at. Would it be better for the dt files to be grabbed from the firmware repo rather than as a subfolder of docs - means we don;t have to keep things updated in two places.

6by9, 2017-03-06:
I think I'm happy to close. The blobs are now updated and work.

There is still a mismatch on the wiring between camera and (display+camera) in GPIOs 2&3 vs 4&5 for the camera control GPIOs, but I didn't sort out the picture updates (taking decent pictures is hard work).  Reopen if you feel that is necessary.

Location of these blobs isn't an issue, and it is better to be part of the docs alongside the text describing them. The master dt-blob.dts as supposedly built into the firmware (it's out of date at the moment) does reside in the firmware repo - Dom has magic scripts that should poke him to consider updating it should the vc4 tree copy change (not automagic to allow for product launch obfuscation!).

closed 2017-03-06 by 6by9