raspberrypi/documentation #501: Update to Mass Storage Device boot instructions

leepenney, 2017-01-24:
Minor quibble, but running through the instructions on booting from a mass storage device in /hardware/raspberrypi/bootmodes/msd.md I found it could use the line:

Type `quit` to exit parted.

after the parted print instructions and before creating the boot and root file systems.

mkretzschmar, 2017-02-08:
Another quibble:

Before "First, prepare the /boot directory with experimental boot files:" there should be a hint that system clock (date/time) should be verified, e.g. "Make sure that the system time is set correctly".
Reason: If the date is wrong (e.g. 2016-11-xx) an curl certificate validation error occures ("curl: (60) SSL certificate problem...") while rpi-update.

JamesH65, 2017-03-06:
We are happy to accept a PR for this change if you have time @leepenney @mkretzschmar

leepenney, 2017-03-07:
Created PR 560 https://github.com/raspberrypi/documentation/pull/560

JamesH65, 2017-03-08:
PR merged, closing

closed 2017-03-08 by JamesH65