raspberrypi/documentation #505: Suggest piping PV in the Linux installation?

aleksandar-todorovic, 2017-02-02:
So, `dd` is a bitch that doesn't show any progress what so ever. _But_, there's a package called `pv` which is more than useful. So, while `sudo dd if=raspbian.img of=/dev/whatever` might not show any output, `dd if=raspbian.img | pv | sudo dd of=/dev/whatever` will (note that `dd` doesn't require root privileges until you get to the `of` part).

So, it might be useful to add a note on that to the Linux installation guide? Seems to me like that's _way_ easier than checking the progress using the `pkill` command.

There's only one caveat, which is why I'm submitting this as an issue instead of doing a PR right away: On Ubuntu-based systems, `pv` is not available by in the default installation, but a simple `sudo apt install pv` does the trick, and the package is less than 100 KB in size and doesn't contain any other dependencies.

Screenshot:

[image]

lurch, 2017-02-02:
While this doesn't _quite_ fall into the same category as #461 I guess it makes the `dd` command more complex, and _might_ lead to more accidental typos? Also, should this retain the `bs=4M` parameter? (and should that be on the 'input' dd or the 'output' dd?)

bennuttall, 2017-02-02:
We get suggestions for this sort of thing all the time - and while it's nice to offer more advanced use cases showing progress and whatnot, they go beyond the scope of *the simplest method of installing an image*, which is what the docs page is for.

Perhaps we could consider an "advanced" page for progress checking versions and whatnot, but we shouldn't add in complexities like this.

Thanks for the suggestion though - it'll come in handy if we do create an advanced page!

bennuttall, 2017-02-02:
I actually use pv a lot myself - but never thought of using it for dd - will remember for next time!

closed 2017-02-02 by bennuttall

aleksandar-todorovic, 2017-02-02:
I highly doubt using `pv` is in any shape of form more advanced than using `pkill` in a different window and then checking for the progress in the `dd` window + making a disclaimer about the buffering. For example, the current version of the installer says:

> The `dd` command does not give any information of its progress and so may appear to have frozen; it could take more than five minutes to finish writing to the card. If your card reader has an LED it may blink during the write process. To see the progress of the copy operation you can run `pkill -USR1 -n -x dd` in another terminal, prefixed with `sudo` if you are not logged in as root. The progress will be displayed in the original window and not the window with the `pkill` command; it may not display immediately, due to buffering.

...which could be simplified with something like:

> The `dd` command does not give any information of its progress and so may appear to have frozen; it could take more than five minutes to finish writing to the card. If your card reader has an LED it may blink during the write process. If you want to run `dd` and see the progress, you might want to install the package called `pv` from your default package manager and then use `dd if=raspbian.img bs=4M | pv | sudo dd of=/dev/whatever". That way, you'll be able to see the progress.

But if you think that's more complicated than the current version of the documentation, it's completely understandable and I respect your decision. :)