raspberrypi/documentation #513: Unable to connect seven inch touchscreen display to compute module 3 IO board

piwizards, 2017-02-20:
Purpose: - Just to use Official 7-inch screen as display 

Using the following hardware:-
1) Compute Module 3 Lite 
2) Compute module IO board V3.0
3) Official 7-inch touchscreen display 
4) 22 way to 15 way DSI display adapter
5) Official 5v power adapter

Software Info:-
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian

Followed the following process to connect:-
1) After running vcgencmd version, received the following output:-
Dec  9 2016 15:11:26
Copyright (c) 2012 Broadcom
version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)

2) Connected the display to the DISP1 port on the Compute Module IO Board through the 22W to 15W display adaptor.
Connected these pins together with jumper wires:
GPIO0 - CD1_SDA
GPIO1 - CD1_SCL
Executed the following command on the Compute Module 3 Lite:-

sudo wget https://goo.gl/Ah6XD5 -O /boot/dt-blob.bin

Rebooted for the dt-blob.bin file to be read, but the screen is blank.
Also supplied 5V extra power supply to PWR IN port of the touch screen.
But still, the screen is blank.

[image]

LizUpton, 2017-02-20:
This is not the appropriate place for support queries; please visit our forums at www.raspberrypi.org/forums if you need help.

closed 2017-02-20 by LizUpton

piwizards, 2017-02-20:
Thanks for your prompt response, but I have posted only in the issues section.
Anyhow, I will also post in the forum link shared by you.

pelwell, 2017-02-20:
1. The Issues section of a repo is to report problems (bugs) in the repo.
1. This is the Documentation repo. Your problem isn't with the documentation, except in the sense that there isn't a document telling you what you are doing wrong.