raspberrypi/documentation #515: USB boot failure when USB camera attached

pishotta, 2017-02-20:
I apologize for posting this here, but my search for a solution leads me to this site.  My problem seems a little more arcane than the RPi Forums can handle.

I followed instructions from

(https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) 

to get RPi-3 to boot from a USB stick (a SanDisk Ultra 64) under updated and upgraded Raspbian Jessie. It works just fine, exactly as described, except it won't boot when I have either of my USB Logitech webcams plugged in.  Sounds like a timing problem, but...

Notes and things I have tried:

1. I added "rootdelay=10" to /boot/cmdline.txt
2. I used the correct "root=PARTUUID=xxx-xxx..." syntax in cmdline.txt
3. I added "program_usb_timeout=1" to config.txt
4. I used the UUID of the /root file system in fstab instead of the /dev/sda2 designation
5. It boots OK with keyboard, mouse, an additional USB hub attached, and/or other additional memory sticks in the USB sockets
6. Using a powered USB hub to connect the camera (in case a power issue) didn't help
7. Juggling what's plugged into what USB slot doesn't help either
8. Neither dmesg, syslog, nor messages show anything obvious (it probably never gets that far)
9. The webcams work OK if I plug either in after the boot completes

But absolutely nothing happens on the screen when a webcam is plugged in (trying two different ones). I get one flash of the activity light on the USB boot device, but nothing thereafter. Works perfectly without a camera plugged in. Seems like specifically the camera is fouling up the hunt for a boot device.

cmdline.txt

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=0065dda2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes elevator=deadline rootwait rootdelay=30

blkid

> /dev/sda1: LABEL="BOOT" UUID="FFD8-3B23" TYPE="vfat" PARTUUID="0065dda2-01"
/dev/sda2: UUID="75af61a7-08a9-4e74-aae5-570cd4c2072e" TYPE="ext4" PARTUUID="0065dda2-02"

fstab

<pre>proc               /proc     proc     defaults     0     0
# /dev/sda1        /boot     vfat     defaults     0     2
UUID=FFD8-3B23     /boot     vfat     defaults     0     2
# /dev/sda2        /         ext4     defaults,noatime   0   1
UUID=75af61a7-08a9-4e74-aae5-570cd4c2072e   /   ext4   defaults,noatime   0   1
# a swapfile is not a swap partition, no line here
# use  dphys-swapfile swap[on|off]  for that
</pre>

lsusb

> Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 0409:0058 NEC Corp. HighSpeed Hub
Bus 001 Device 006: ID 0781:5583 SanDisk Corp.
Bus 001 Device 007: ID 046d:0809 Logitech, Inc. Webcam Pro 9000
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I note that "lsusb -v" does show the webcam to be the most complicated device with gobs of interfaces.

Like I said, looks like the presence of the usb webcam is either slowing or blocking the finding of the bootable usb mass storage device 006, but I'm out of ideas as to how to deal with this.  AFAIK, it is only the presence of a webcam that causes this to happen.  Perhaps there are additional options?  At the very least, I thought I should bring it to your attention so that if this is a systemic issue and there is a solution to what may become a more common problem, it can be referenced in the documentation.

-- Fred P.

thegnnu, 2017-02-20:
I do not know if this will help but I had week's of trying to get 2 Hard disk & 1 USB Pen to boot reliably until I put the following before the UUID and   added rw, in the fstab so your would look like this.
/dev/disk/by-uuid/75af61a7-08a9-4e74-aae5-570cd4c2 / ext4 rw,defaults,noatime 0 1
/dev/disk/by-uuid/FFD8-3B23 /boot vfat defaults 0 2072e / ext4 rw,defaults,noatime 0  1
not =75 and =ff has also changed UUID gone to uuid. I know the defaullts contain the rw but it would not boot as it showed a read only until i added rw before the defaults. sorry not a expert just a user who struggled to get 3 usb media drives to work. 
TerryR

lurch, 2017-02-20:
A better place to post this is probably at https://github.com/raspberrypi/firmware/issues ?

pishotta, 2017-02-20:
Thanks, thegnnu.  I'll investigate further and try fiddling with those changes to fstab, but I don't think it even gets to the point where it can read fstab.  By the way, I also have an RPi-3 with two mirrored hard drives and other USB gizmos hanging off of it (operating as a file and media server), but I did not experience your problems setting it up.  Got lucky there.

And lurch, thanks for the suggestion.  I re-posted there.  Should some solution be forthcoming from that source,  I'll follow up here (and vice versa).

-- Fred P.

closed 2017-02-22 by ghollingworth