raspberrypi/documentation #522: RPI 3 Wifi&Bluetooth Connection

mbt28, 2017-02-28:
Hi,

I am building a baseboard for CM3, for the sake of compatibility and simplicity I would like to use bcm43438 on my board. 

I am curious will you ever planning to release the schematics of wifi combo module?
How BT part is connected? How the pcm data communication made?

Thanks

MatsK, 2017-03-08:
Isn't this a topic for the forum, like this one  [Compute Module](https://www.raspberrypi.org/forums/viewforum.php?f=98)

JamesH65, 2017-03-09:
Yes, really a forum topic, but for completeness there is no intention to release the schematics of the Wireless/BT module.

closed 2017-03-09 by JamesH65

marcosscriven, 2017-06-17:
@JamesH65 - I'm trying to determine whether the BCM43438 on the RPi 3 has the correct GPIO pins connected to support Bluetooth "Triggered Clock Capture". The product declaration on the Bluetooth website seems to suggest it does: See rows 24 and 25 of table 5 [here](https://www.bluetooth.org/tpg/showCorePICS.cfm?3A000A5A005C5346565A5D12136B4F1D17270F0812080D34405C210C0D130B285C44)