raspberrypi/documentation #559: Enable SSH on headless Raspberry Pi's

MatsK, 2017-03-07:
On https://www.raspberrypi.org/documentation/remote-access/ssh/

I would propose that the last part gets a heading like:
4. Enable SSH on headless Raspberry Pi's

to highlight this feature.

JamesH65, 2017-03-08:
Not sure about this. The headless stuff is covered, albeit without a title. It really needs to go in that section 2 to keep the order correct. Perhaps just a better highlighted title? Dunno. @AisforAstronaut @LornaLynch  any comments?

AisforAstronaut, 2017-03-08:
Another sub heading could work!

JamesH65, 2017-03-08:
@AisforAstronaut  Do you want to give it a go?

AisforAstronaut, 2017-03-08:
*Rolls up sleeves*

AisforAstronaut, 2017-03-08:
Edited in GitHub, but I have no idea how to publish that to the website.

bennuttall, 2017-03-08:
It deploys automatically every hour

AisforAstronaut, 2017-03-08:
A proper magic trick, then! Thank you! I shall wait for the fairies to do
their jobs...

:-)

JamesH65, 2017-03-08:
Fixed, closing.

closed 2017-03-08 by JamesH65

adsche, 2018-10-22:
What are the credentials when setting up ssh like that? Should that be added to the documentation?

JamesH65, 2018-10-24:
Default credentials are just the standard logins.