raspberrypi/documentation #567: config-txt.md minor changes, suggestion.

MatsK, 2017-03-08:
[https://www.raspberrypi.org/documentation/configuration/config-txt.md](https://www.raspberrypi.org/documentation/configuration/config-txt.md)

The section: Monitoring temperature and voltage
is mixing a how to meassure voltage manually and how to get the temperature from CLI/Bash.

Suggestion is to split them into two sections, "Monitoring temperature" and "Meassuring PSU voltage manually".

JamesH65, 2017-03-08:
This section has very recently (today) been reformatted - can you point me at the new file the issue is in? The link  provided is now broken.

MatsK, 2017-03-08:
OK,
[https://github.com/raspberrypi/documentation/blob/master/configuration/config-, stxt/overclocking.md](https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/overclocking.md)

JamesH65, 2017-03-09:
I have made a change to the appropriate file and also added a bit more new information #569. Please check out and comment.

JamesH65, 2017-03-15:
Fixed by #569

closed 2017-03-15 by JamesH65