raspberrypi/documentation #572: documentation URI in /boot/config.txt leads to 404

jjsanderson, 2017-03-10:
I'm not entirely clear where `/boot/config.txt` is generated; this may not be a documentation bug as such. However, on a fresh flash of 2017-03-02-raspbian-jessie.zip, the header comment of that file points to: `http://www.raspberrypi.org/documentation/configuration/config-txt.md`.

...which leads to a rewritten address and a 404 error.

The correct link should presumably be `https://www.raspberrypi.org/documentation/configuration/config-txt/README.md`.

Trivial nit-picking error, I know, but this is the sort of thing that derails newcomers. Or, indeed, people with misplaced confidence that they may know what they're doing. Like... er... me.

JamesH65, 2017-03-10:
The config.txt documentation has moved slightly, two days ago! I had no idea this was referenced from the config file itself. Will get that updated asap.

XECDesign, 2017-03-10:
Well, spotted. I can make sure the next image points to the updated location. Perhaps a goo.gl link should be used to prevent this happening in the future.

JamesH65, 2017-03-10:
It's not likely to change again in the foreseeable future, but having links like that are always going to be prone to becoming out of date. There's no way when updating the documentation to know they even exist outside the documentation tree.

XECDesign, 2017-03-13:
Would you prefer a goo.gl link or just a direct link?

jjsanderson, 2017-03-13:
Doesn't a goo.gl link just provide another point of failure? – in that somebody would have to remember to update the link address if the document ever moved, and that person would need access to the account, and so on... or have I misunderstood?

XECDesign, 2017-03-13:
I believe the address the link redirects to can be changed without having to change the link itself. Although I can easily fix the link on future installs, config.txt is a little trickier to update on existing installs. Right now the solution is: update a Raspbian package and make it update the file when it's upgraded, which is a little more prone to go wrong than just updating the redirection.

That's still not much use if we forget who owns which goo.gl link, but that should be easy to sort out with git blame.

XECDesign, 2017-03-13:
Should be fixed by https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/7c9e35de87fa5392fd01ac4961d877aa20ee4fcc and https://github.com/RPi-Distro/pi-gen/commit/9cffa42a3514cd2a4d0aa648a6e52860ae86e1b5

JamesH65, 2017-03-14:
@XECDesign has fixed this so closing. Reopen if further issues arise.

closed 2017-03-14 by JamesH65