raspberrypi/documentation #574: No instruction on ho wto change screensaver/blanking in Pixel

JamesH65, 2017-03-11:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=177001

I beleive this is non-trivial so we could do with some instructions.

@spl237 Any thoughts?

JamesH65, 2017-04-25:
@spl237 Happy to write the page if I know what to write! Which I currently don't, but am eager to learn...

spl237, 2017-04-25:
To be honest, I'd have to look up what to do, as I've never changed it either...

bennuttall, 2017-04-25:
+1.

I've been asked how to do this a handful of times - and I've looked it up or asked around and found the solution but never remembered what the command is. Would be really handy to have it somewhere I can look up, that one time you want to keep the screen alive when it thinks it's idle.

JamesH65, 2017-04-25:
OK, been looking in to this - think I know what needs to be documented. I'll get on with it.

JamesH65, 2017-04-25:
Does this fit the bill?

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-1%23574/configuration/screensaver.md

JamesH65, 2017-04-25:
Actually, I'll PR it, any comments can go on that.

JamesH65, 2017-04-25:
Fixed by #627

R4C3R, 2017-04-25:
just for info:

In order to disable screen blanking of command line interface I use the
following

Edit /etc/kbd/config and make sure these values are set as follows in order
to prevent that the screen enters in screen save / power save mode:

sudo nano  /etc/kbd/config

BLANK_TIME=0

POWERDOWN_TIME=0

In JESSIE there is a bug that prevents the above options being set, you
have to edit

sudo nano /etc/init.d/kbd

Change

   # screensaver stuff

   setterm_args=""

   if [ "$BLANK_TIME" ]; then

       setterm_args="$setterm_args -blank $BLANK_TIME"

   fi

   if [ "$BLANK_DPMS" ]; then

       setterm_args="$setterm_args -powersave $BLANK_DPMS"

   fi

   if [ "$POWERDOWN_TIME" ]; then

       setterm_args="$setterm_args -powerdown $POWERDOWN_TIME"

   fi

   if [ "$setterm_args" ]; then

       setterm $setterm_args

To

   # screensaver stuff

   setterm_args=""

   if [ "$BLANK_TIME" ]; then

       setterm_args="$setterm_args -blank $BLANK_TIME"

   fi

   if [ "$BLANK_DPMS" ]; then

       setterm_args="$setterm_args -powersave $BLANK_DPMS"

   fi

   if [ "$POWERDOWN_TIME" ]; then

       setterm_args="$setterm_args -powerdown $POWERDOWN_TIME"

   fi

   if [ "$setterm_args" ]; then

       #setterm $setterm_args

       TERM=linux setterm > /dev/tty1 $setterm_args

Maybe this helps

Rolf Bakker
Lerchenweg 14
56075 Koblenz
Deutschland

m: +4915755211456
@: rolf.bakker.70@gmail.com

JamesH65, 2017-05-08:
Fixed by #627

closed 2017-05-08 by JamesH65