raspberrypi/documentation #599: Documentation required - VPN

JamesH65, 2017-03-24:
It would be useful to have some information on how to set up a VPN on the Pi.

karanssj4, 2017-04-01:
would a post on how to use openVPN via PiVPN be sufficient? @JamesH65

JamesH65, 2017-04-03:
I'm not thought about it much yet - just put in the issue as a placeholder
to remind me to look at it at some point. If you have a set of docs that
could sent in as a PR, that would be a great start.

karanssj4, 2017-04-03:
okay, i'll take this up

JamesH65, 2017-04-25:
@karanssj4 Did you get anywhere with this?

karanssj4, 2017-04-26:
oh i almost forgot about this! :P
i used piVpn successfully, i'll make a doc for tutorial by this weekend most probs

JamesH65, 2017-05-17:
ping

JamesH65, 2018-08-20:
We are happy for people to provide PRs, which will be technically verified before merging.

adeeb10abbas, 2018-08-20:
Okay I am on it!

JamesH65, 2019-01-18:
Any of the prospective contributors manage to knock anything up?

adeeb10abbas, 2019-01-21:
> Any of the prospective contributors manage to knock anything up?

Hey there, I completely forgot about this! I found an article on Free Code Camp's Medium page which demonstrates something very similar to what I did.
Here's the link-> [https://medium.freecodecamp.org/running-your-own-openvpn-server-on-a-raspberry-pi-8b78043ccdea](url)

epoch1970, 2019-09-07:
VPN, "vaste programme..." To make progress on this I think we need scenarios. 2 legitimate use cases I see:

- "Site-to-site". OpenVPN does that fine, the config is simple. One Pi per site (good!). Can work routed or bridged (my preference, true "telepresence").
Could even help out people who want to bridge a wireless client -an old post on that [here](https://www.raspberrypi.org/forums/viewtopic.php?t=204361) that has had a remarkably small uptake ;)

- "Roadwarrior", e.g. dorm-to-home. Here, OpenVPN "server mode" requires TLS  and explaining what a PKI is and how to manage it is, IMHO, beyond hope. This scenario  could be done with WireGuard (it uses symmetric keys), although its routed nature will require a bit of extra explaining. 
In either case, the client side install guide might need to take into account client applications on such and such platform. Can that be avoided by locking down the doc to a single platform, like Raspbian Desktop?

Now, I don't think I take much risk in saying the vast majority of the people interested in a VPN want in fact to circumvent geoIP restrictions from commercial services; They would read those docs and complain loudly they're useless. Then the tin-foil hat crowd would complain Pi doesn't include hardware crypto and the demonstrated setups are "dangerously weak".

The project might also clash with PiVPN and other "VPN made simple" initiatives. Not that I would mind, personally.

I'm subscribing to this thread, just in case.

lurch, 2019-09-07:
I know very little about VPNs, but replying to just:

> Can that be avoided by locking down the doc to a single platform, like Raspbian Desktop?

Yes, all the documentation in this repo is written exclusively for (the latest released version of) Raspbian.

aallan, 2020-06-03:
There are two (three?) possible documentation needs here

1A) Setting up a VPN server on your own Raspberry Pi so you can VPN "home"
1B) Setting up a point-to-point tunnel between two sites via a VPN
2) Using a VPN client on the Raspberry Pi to connect to a commercial or work VPN. 

There is a lot of third-party documentation around setting up a VPN server, although it might be useful to have a basic guide in our own documentation. However that runs into the whole problem of dealing with people's home routers (firewall holes etc) and not-fixed IP addresses that home connections often have to cope with (and so DDNS is needed).

I think point-to-point tunnels are beyond the scope of what our documentation should cover. If you have two sites with two separate LANs, making then one big VLAN using a VPN between them sounds like something you should probably already know how to do.

I think the biggest documentation need is using the Raspberry Pi to *connect* to a VPN rather than act as a server. There is very little third-party content around that. I wrote [a blog post](https://www.raspberrypi.org/blog/working-from-home-with-your-raspberry-pi/) recently for the Pi blog that covered some of the basics, possibly we could recycle this for the documentation?

JamesH65, 2020-06-03:
Sounds sensible. Attaching to a VPN is the biggest use case for Pi's I would think, although there is an argument that quite a number of people might be using the PI as a gateway in to their own network, where using it as a VPN server would be appropriate. Along with things like PiHole.

epoch1970, 2020-06-03:
The more I think about the more I think it's the wrong idea. (not to disparage the blog post, I just read it and it is impressive.) Ppl want to connect to for-pay VPN servers to watch N*tflix from a region different from the one they contracted for. Ok, not all people, only 98% of them.

VPN is on the way out. Some overlay service ala zerotier is in. IPv6 is in. 

Managing an incoming connection is a nightmare. A "meeting point" in the cloud (=Pis making an outgoing call to the mothership) could be perhaps something for somedomain.io run by RTP/RPF. Discovery/membership is IMHO the crucial service. 

What happens next is a matter of user preferences and applications. There are useful applications indeed.

aallan, 2020-06-03:
Okay, I'll put a page on "joining a VPN" on my to do list. Although I think integrating better VPN client support into the OS should be on the list of changes. It really should be a GUI-only thing like joining a WiFi network. Having to drop to the command line to do something like that isn't great.

lurch, 2020-06-03:
> Discovery/membership is IMHO the crucial service.

There used to be services on https://github.com/raspberrypi/documentation/blob/master/remote-access/access-over-Internet/README.md#alternatives that offered that kind of thing, but that section got removed because it was too difficult to try and keep up to date.
(And I agree with @JamesH65 that trying to keep the number of external links in the documentation to a minimum is a good thing!)

epoch1970, 2020-06-03:
Ah yes I saw dataplicity in an older revision of the file. Much like what I was thinking of, except that actors in this category, IMHO _i._ are not you (user trust issue, visibility/enrolment issue) and _ii._  have to get money from somewhere and that ends in marketing puff/api bloat.

aallan, 2020-06-04:
I'm adding VPN content on the [``aallan-vpn-docs``](/raspberrypi/documentation/tree/aallan-vpn-docs) branch.

aallan, 2021-06-07:
I really need to either drop this or commit the branch due to #1911.

aallan, 2021-06-17:
I'm going to add the content for this in post-#1911 so closing for now.

closed 2021-06-17 by aallan