raspberrypi/documentation #600: Documentation required - Security hardening

JamesH65, 2017-03-24:
A frequent request on the forums is for help on configuring the Pi to make it secure. For example, setting up fail2ban, or SSH key rather than password connections. Probably lots of other things that could be added.

sschadwick, 2017-03-24:
Here's a list of what I generally do when I init a raspi:

- `sudo apt-get update/upgrade`
- `sudo apt-get install apt-transport-https`
- Change default username/password
- Ssh-copy-id
- fail2ban
- UFW or Iptables
- Enable ssh and change default port

XECDesign, 2017-03-24:
apt-transport-https will be preinstalled on the next image (if it wasn't already ont he exist one), but it does not add any security. Apt already verifies that package lists are signed and package checksums are correct.

I'd also argue that changing the SSH port does nothing for security. Instead, disabling password logins entirely would be the way to go.

The google-authenticator pam module could be added, but it's probably too much.

JamesH65, 2017-03-24:
Currently working on a draft security page,

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-1-security/configuration/security.md

Will be adding ssh keys, fail2ban.

XECDesign, 2017-03-24:
A few observations which may or may not be useful:

Unfortunately, there are still bits of Raspbian which have the pi user hard coded in, so some things won't work properly if that user is removed.

There are also bits which require passwordless sudo, which can be considered a security problem worth pointing out.

pkexec provides similar functionality and is also configured to allow the pi user to run commands as root without a password.

Use of visudo can overlook the changes in /etc/sudoers.d/* unless the exact file to edit is passed using the `-f` flag (for example, `sudo visudo -f /etc/sudoers.d/010_pi-nopasswd`)

JamesH65, 2017-03-25:
Thanks @XECDesign, all useful things. I'll keep adding to the doc until
it's felt it is sufficient.

JamesH65, 2017-03-28:
I've made a number of substantial changes to the draft if anyone fancies a look. Link as above.

XECDesign, 2017-03-28:
`cat id_rsa.pub >> ~/.ssh/authorized_keys` is likely to get permissions wrong if the file does not already exist. I don't know if this is the case in Raspbian, but some versions of SSH ignore the file unless it has correct permissions.
From the client, you can run ssh-copy-id pi@ip to copy the key properly. In the past, there were some complaints about it, since windows and mac users might not have immediate access to the command, but the same goes for ssh-keygen.

Maybe I missed it, but I don't see a mention of disabling plain text password logins. Only adding the key would still leave password logins wide open.

bennuttall, 2017-03-28:
`ssh-copy-id` is a great tool for this. It was suggested to be added to the SSH page but someone pointed out it isn't available (or installed by default) on Mac.

I'd add "Disable password authentication in sshd config" to the list, so that it only allows ssh key authentication, no passwords.

Also: keep your Pi's packages up to date by regularly upgrading them. And Pete from Mythic Beasts recommended to me recently to install the `ssh-server` package explicitly so it gets updated with system upgrades.

XECDesign, 2017-03-28:
@bennuttall do you have more information on why openssh-server wouldn't be upgraded normally?

bennuttall, 2017-03-28:
He said because it's not explicitly installed, it wouldn't be. But now I think about it, perhaps he was thinking about packages which are brought in as dependencies, which may not be. I'll try to bring him in...

mythic-beasties, 2017-03-28:
I suggested putting into cron every day,

apt-get install openssh-server

so that way the ssh server gets updated very quickly with minimal fuss even if you're a bit slow about installing updates in general and don't want to auto-update everything.

Our Pi3 cloud does apt-get -y upgrade, nightly by default on the assumption that people aren't buying managed services or subscribing to the debian security update list and pro-actively applying updates.

mythic-beasties, 2017-03-28:
fail2ban doesn't currently support IPv6, sshblack can be a better choice. Similarly if you're configuring ufw make sure it does iptables & ip6tables. Bear in mind that IPv6 link-local means that you're listening on your local lan even if you think you're not on a v6 capable network. Lastly, if you're on a netbooted server, take a lot of care with ufw or you'll firewall out your fileserver.

bennuttall, 2017-03-28:
Ah, running `apt-get install openssh-server` to upgrade the package makes sense.

I've never put an upgrade in cron (though I'd like to automate it) in fear of the upgrade asking a question about varying versions of files. The `-y` flag doesn't deal with this. I think another flag, or the no frontend thing might do, but I don't really want to choose between always taking the package's version, or always keeping mine.

XECDesign, 2017-03-28:
Ah, thanks for clarifying. For a second there I had an "everything I thought I knew is wrong" moment.

mythic-beasties, 2017-03-28:
Automated upgrades are a big argument, basically it depends on how far from stock config you go, how important the server is, how good your manual patch management is. We don't do it for our managed services, but we read security announcements and can patch ~ 1000 servers in less than half an hour. That's a bit different to a hobbiest Pi3.

bennuttall, 2017-03-28:
@XECDesign I had a similar feeling when Pete told me. I think I misunderstood though.

JamesH65, 2017-03-28:
Hmm, I'm no expert of security or installing a lot of this stuff, but there
is clearly a need to have some sort of docs page on security. The doc so
far is simpy me trolling arouddn the net finding what people have been
doing and collating it.

I think others are going to have to fill in some blanks in the draft so
far. I'm happy to accept PR's in to the draft branch should anyone have the
time and inclination to fill in those blanks. Or even just suggest where
the blanks are, and what they should be filled with. I can them research
and do the heavy lifting, but others will need to proofread!

James

bennuttall, 2017-03-28:
Happy to help - I'm sure we can put something together, with Pete's suggestions and approval :)

JamesH65, 2017-03-29:
@mythic-beasties A check google shows that fail2ban does have IPv6 support once you get to v0.10. However the one installed as standard in Raspbian appears to be 0.8.13 (really quite an old version). 

Pete, would you be happy to knock up some docs on some of the other stuff you have mentioned? We don't need to go mad, just basic security stuff.

mythic-beasties, 2017-03-29:
So a default configuration with

(i) openssh open to the internet auto-updating
(ii) a restrictive firewall limiting access to only openssh
(iii) sshblack/fail2ban
(iv) only key based login
(v) an explanation of how to open up a port for $my_new_service

I'll see if I can get some time to write this.

JamesH65, 2017-03-29:
Check the current draft documentation first, don't want to do the work
twice!

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-1-security/configuration/security.md

Hacksawfred3232, 2017-04-24:
Sorry if late. While I am getting my problem concerning USB boot fixed, I thought I might add to this.
When I am running my server, I am mainly concerned about SSH. Just in case someone did get into my network via some vector I have not noticed, I have locked out all of the users except for my account. The account has a strong password on it so they cant get in via that. 
There is a option in sshd that can do this. Here is a link to it. 
https://www.cyberciti.biz/tips/openssh-deny-or-restrict-access-to-users-and-groups.html

JamesH65, 2017-04-24:
Thanks, I have added appropriate content.

bennuttall, 2017-04-24:
Some notes you may find useful in the "SSH gateway" section of this article I wrote: https://opensource.com/article/17/4/5-projects-raspberry-pi-home

mythic-beasties, 2017-04-28:
>Some notes you may find useful in the "SSH gateway" section of this article I wrote: https://opensource.com/article/17/4/5-projects-raspberry-pi-home

I think the fail2ban/sshblack recommendation is a bad idea. The Pi will always see the source IP address as that of the NAT gateway so failed password attempts will block the gateway and all external traffic.

If you've got native IPv6 and see real source v6 addresses it's a good idea though.

Pete

-- 
Mythic Support <support@mythic-beasts.com>
Live service status: http://status.mythic-beasts.com
Follow us on Twitter: https://twitter.com/Mythic_Beasts

JamesH65, 2017-06-05:
@mythic-beasties Pete, would you mind giving the draft a once over?  Not sure what may or may not need to be added. 

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-1-security/configuration/security.md

mythic-beasties, 2017-06-06:
Looks pretty good to me. If you've disabled passwords fail2ban/ufw add minimal security versus a denial of service risk, but they're a really good idea if you're still allowing password logins.

JamesH65, 2017-06-06:
Thanks Pete,

It's only meant as a primer, so I think it good enough to get people
thinking about security. I'll push it forward for copy editing and addition
to the website.

James

JamesH65, 2017-06-06:
PR #661 sent to copy edit.

JamesH65, 2017-06-08:
Fixed by #661

closed 2017-06-08 by JamesH65

epoch1970, 2019-08-26:
The value of fail2ban is very much overstated in the document. The reality is that this kind of “active” security devices is a DoS attack vector, and often self inflicted. I don’t think it belongs in a security primer.

If securing password-based installs is at stake, then I would instead suggest a section on [google’s OTP PAM module](https://github.com/google/google-authenticator-libpam/)