raspberrypi/documentation #611: Website Font for 'code' text

bsimmo, 2017-04-07:
l and 1 in text are very similar in the 'code' part.
Consider using something easier to distinguish.
Maybe look at dyslexic and teaching fonts, they should know what is best for the situation.

I know I've had it before and it just came up from two people on the new AP article.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=179618

JamesH65, 2017-04-10:
@bennuttall Ben, is there anything we can do about this?

bennuttall, 2017-04-10:
I don't think so. There's always issues telling `1` from `l` in fonts, especially monospace. Code needs to be a monospace font. Unless someone suggests a monospace font that is considerably better, then we can't really do anything.

bsimmo, 2017-04-10:
There are some that use a curly l, a quick search shows quite a few here. 
No idea on the implications of getting them in use on a website or if PIXEL may need them as well.

https://fedoramagazine.org/5-great-monospaced-fonts-for-coding-and-the-terminal-in-fedora/

XECDesign, 2017-04-10:
Hack tends to be pretty good.
https://github.com/chrissimpkins/Hack#web-font-usage

JamesH65, 2017-04-25:
So, should we use hack? If so, how do we use hack? Does the font need to be on the end users device?

bennuttall, 2017-04-25:
@grega do you have any comments on this?

XECDesign, 2017-04-25:
Looks simple enough to use without having it installed:
https://www.w3schools.com/code/tryit.asp?filename=FEZ96O3WIJQV

JamesH65, 2017-06-06:
@grega Is there any chance we could do this? Not my field...

grega, 2017-06-07:
@JamesH65 yep this is definitely possible to sort out, I've added this to the current support backlog & will confirm the font choice with the design team.

JamesH65, 2017-11-20:
@grega Hi Greg, ISTR this being fixed? If so can this be closed?

grega, 2017-11-30:
@JamesH65 It's fixed:

[image]

closed 2017-11-30 by grega