raspberrypi/documentation #615: New remote access method

remoteiot, 2017-04-12:
Hi Sirs, 

In the remote-access README.md, could we add a new remote access method?

The page is https://github.com/raspberrypi/documentation/blob/master/remote-access/access-over-Internet/README.md

The content is attached.

[RemoteIoT.txt](https://github.com/raspberrypi/documentation/files/915607/RemoteIoT.txt)

Thanks very much for your attention,

Best regards,
David

bennuttall, 2017-04-12:
If it's only a free trial, I don't think it's suitable for inclusion.

remoteiot, 2017-04-12:
Hi Ben,

It is not only a trial. It is a good solution as the remot3.it. 

Best regards,
David

JamesH65, 2017-05-12:
Made the decision to not incorporate this change. I'm also considering removing a lot of the details providing on the original page, and just having a list of links possible alternatives (which would include this one). We are not an advertising platform for any of these applications.

closed 2017-05-12 by JamesH65