raspberrypi/documentation #620: Boot from USB instructions not working as referring to dev/mmcblk0p2 instead of partuuid

keunes, 2017-04-14:
Not sure if the same issue as in #434, so therefore creating a new issue.

I today downloaded the latest Raspbian Jessie Lite and wrote it to my micro-SD card. Followed all instructions on the page mentioned below, but after removing the SD card I got an error.
It appears that in the documentation, cmdline.txt and fstab should be adjusted according to

`sudo sed -i "s,root=/dev/mmcblk0p2,root=/dev/sda2," /mnt/target/boot/cmdline.txt`
`sudo sed -i "s,/dev/mmcblk0p,/dev/sda," /mnt/target/etc/fstab`

In my installation, by default, however, the system is set  up with partuuid. Consequently, the code proposed on the help page doens't work to change the boot and root locations, and as such boot fails after removing the SD.

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

XECDesign, 2017-04-14:
See https://github.com/raspberrypi/documentation/pull/616

JamesH65, 2017-04-19:
New docs merged.

closed 2017-04-19 by JamesH65