raspberrypi/documentation #640: How to make changes to .BASHRC AND .BASH_ALIASES work

jasonmerry, 2017-05-14:
I am getting more and more familiar with using a shell due the use of node.js and such like for building websites, so most of what I have read in your docs makes a lot more sense to me, more now than it would have done several years ago.

Brushing up and taking a new fresh look at a beginners guide I thought would help more of it stick in the grey matter. However, after following the helpful guide about .bashrc and creating aliases I was unable to get them working.

I was sure there was something else I needed to do but didn't know what so I had to search for it. I found what I was looking for and pretty quickly as I sort of knew what I was looking for. Something I don't think a complete noob would know how to search or even should have to search for. 

I think it would be worth adding the last step of editing a .bash file to the end of the .bash guide:

```
source .bashrc
```

JamesH65, 2017-05-15:
can you post a link to the page you believe needs this change?

jasonmerry, 2017-05-15:
Sure :)
https://www.raspberrypi.org/documentation/linux/usage/bashrc.md

JamesH65, 2017-05-15:
OK, that page could do with an update, but the particular thing you are requested might not be the best approach since that reruns the entire bashrc which might not do exactly what you think. Usually you simply start a new terminal to get the changes.

I'll try and make things a little clearer in a rewrite, will be a few days.

JamesH65, 2017-05-15:
Fixed by #641

jasonmerry, 2017-05-15:
Ah, ok. I did not know that about `source` but I did know about the new terminal. 

Thanks for updating :)

JamesH65, 2017-05-16:
Fixed by #641

closed 2017-05-16 by JamesH65