raspberrypi/documentation #649: Incorect explanation of "WHAT THE COLUMNS MEAN"  under "Setting up and apache web server on a Raspberry pi"

N1MKG, 2017-05-19:
Column #2 "The number of files in the directory (of 1 if it's a file)"  is incorrect.
This column is the number of HARD LINKS to the inode (The internal representation of all objects in the
filesystem).  For a file it usually is 1.  For a directory it is the number of places in the filesystem that have
a reference to this directory.  A directory (when created) starts with the number "2".  The parent directory
has a reference (by the directory name) to the inode, and inside the directory it has its own reference
to itself known as ".".  When a new sub-directory is added under this one, the count goes up by one
to account for the ".." reference in the new sub-directory.  You can add/delete the number of files in any
directory and the reference count will not change.

JamesH65, 2017-05-22:
Can you provide a link to the page with this error? I cannot find it.

N1MKG, 2017-05-22:
Here's the link you requested.

https://www.raspberrypi.org/documentation/remote-access/web-server/apache.md

The error is about 1/2 way down the page on the line that starts with "2)".
I have explains (in way too much detail) what the second line is 
referring to.
I AM NOT giving a replacement to #2.  That should be someone else that can
correctly condense my details into a correct sentence or two.

PS. I have no idea why to page author was trying to explain what the columns
     mean.  I would assume anyone reading this page (ie someone that is 
installing
     apache) would already understand the output of a directory listing. 
Since the
     next step is much much harder being to configure the apache web 
server (ie
     all the .conf files), and possibly doing the next section 
installing and
     configuring PHP.

Also,
     1) Just above the line "WHAT THE COLUMNS MEAN", the previous 
sentence ends
         with "/www/"  and should be "/var/www/"
     2) Number 1) "The permissions of the file or directory" should as a 
minimum
         explain "d" refers to a "directory", and the breakdown of 
"rwx", "r-x",
         and "r-x" are the permissions given to "Owner", "Group", and 
"All others".
         And "r" is for read, "w" and for write, and "x" is for file 
"exacutable"
         and directory "searchable".  Any position with an "-" means 
that permission
         is turned OFF>

PPS In case you can't tell, I have spent over 35 years teaching unix, 
linux, and many
     other varieties or operating systems (include system internals).

JamesH65, 2017-05-25:
Thanks for the link, I also cannot see why that section needs to be there at all. I'll do a PR and fix up the page.

JamesH65, 2017-05-25:
PR #653 should fix this.

N1MKG, 2017-05-25:
Looks great!  No extra unneeded info :-)

N1MKG, 2017-05-25:
When will these pages be pushed/fetched from the Raspberry PI web site?

closed 2017-05-25 by N1MKG

JamesH65, 2017-05-25:
It will go to our copy editors for proof reading (indeterminate timescale -
depends on their workload), then I merge it on github, then about an hour
later it automatically gets pushed to the website.