raspberrypi/documentation #650: Bug in documentation regarding wpa_supplicant psk entry?

felixgonschorek, 2017-05-22:
The documentation under https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/wireless-cli.md#adding-the-network-details-to-the-raspberry-pi states, that wpa_passphrase returns a config with unquoted "psk" value and states:

> Note that, although wpa_passphrase returns the psk value unquoted, the wpa_supplicant.conf file requires this value to be quoted, otherwise your Pi will not connect to your network.

This is not true for the raspian i setup today on my Raspberry PI Zero (2017-04-10-raspbian-jessie-lite.zip).

The pi did not connect to my WPA2 secured WiFi network when i added quotes, it only worked without quotes.

Maybe this is outdated information, i am just starting to work with raspbian so i don't know? I can provide a pull request with updated text if the documentation is wrong here.

Thanks!

JamesH65, 2017-05-24:
Hmm, I just checked the default wpa_supplicant file and that uses quotes. Also, all references I can find use quotes, so I am not sure what your issue was but I don't think it was the quotes. Perhaps you were using odd characters that needed 'escaping'?

felixgonschorek, 2017-05-24:
Okay, i dug a little deeper and as i understand it now, it seems, that quoted psk entries means "plaintext" password and without quotes it's the ecnrypted/hashed/whatever version which is being returned by my wpa_passphrase program.

when i run:
`wpa_passphrase TEST123 TEST123456789`
i get in return
```
network={
	ssid="TEST123"
	#psk="TEST123456789"
	psk=7d30370a82e4e39af70ad33a5cdb272887f19e6ab4f0bdd3bdac6960f625704b
}`

also i the (i think it's official wpa_supplicant) documentation at http://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf i could not find any instructions on quoted / unquoted psk values but all examples on the end of the document (marked with "# Example blocks:") use quotes for plain text passwords and the unquoted hexadecimal representation for hashed/encrypted/whatever passwords.

so i think the documentation could reflect that to give people a chance to distinguish. since im not 100% certain if i am right i would be willing to test this out on my pi if you were interested?

JamesH65, 2017-05-24:
Ah, that rings a bell. I'll reopen and have a think.

JamesH65, 2017-05-24:
Hmmm, this was recently changed - but I think the change is incorrect (bad checking by yours truly) a plaintext password should be quoted, the encrypted one should be unquoted. I will double check and do a PR.

JamesH65, 2017-05-25:
I've issues a PR that reformats this page a bit to make it more understandable, and correct the mistake.

Fixed by #652

Once it has been copy edited and merged , I'll close this.

PR #652 (2017-06-05): Update wireless-cli.md
Fixes #650
closed 2017-06-05 by JamesH65