raspberrypi/documentation #651: net_tutorial.md:  copy kernel to tftpboot breaks kernel update

D-Albers, 2017-05-23:
Hi,

the command "cp -r /boot/* /tftpboot" breaks the option to update the kernel through ap-get or rpi-update, because the folder "tftpboot" is not visible from the inside of the running pi.

Possible solution: Create a softlink from "tftpboot" to "boot".

ghollingworth, 2017-05-24:
I'm not sure what you mean by 'running pi' do you mean **server** or **client**

The **server** could export the /tftpboot directory over nfs so that the **client** could mount it by default in `/etc/fstab` this would mean the **client** can update it's filesystem independently of the **server**

This has a disadvantage in that if there are many **clients** using the same tftpboot directory then they will also share at a minimum the `bootcode.bin` file (the other files can be added into `[serialnumber]/start.elf` and added to fstab on the **client** )

Not sure this is something to add to this tutorial though.

D-Albers, 2017-05-24:
> I'm not sure what you mean by 'running pi' do you mean server or client.
With the 'running pi' I referred to the client one.

The problem I see in the guide is, that the client is not able to update the files in `/tftpboot/[serialnumber]/` **without any further work behind the end of the guide**. This is something I didn't see before at the documentation from the Raspberry Pi Foundation. The other guides have been all also usable for the absolute beginners. 
One of the worst cases I may expect, is a changed `config.txt` inside `/boot/` on the shared nfs but not in the `/tftpboot/[serialnumber]/`. It the same with the kernel one, different versions in the `/tftpboot/[serialnumber]/` folder and in the `/boot/` on the shared nfs.

A shared `bootcode.bin` shouldn't be really a problem or did I miss there something?

Sure, mounting the `/tftpboot/[serialnumber]/` on the clients may be also an option, but there is no hint in the guide why it should be done.

I expect, that the target audience of the guide is still the beginners and not the freaks, which are having the necessary knowledge and experience. Is the expectation wrong?

ghollingworth, 2017-06-06:
In the end we can't do everything in a tutorial like this, if you've got a suggested modification to add mounting of the /tftpboot/serialnumber to the documentation we're happy to accept a pull request.  Otherwise this is up to the user to handle themselves.

JamesH65, 2017-06-06:
I'll close then. If a PR turns up it will certainly be considered.

closed 2017-06-06 by JamesH65