raspberrypi/documentation #656: 3,3V regulator + TSOP38 = 55.55 Hz noise

arthurmoises, 2017-06-02:
Didn't found the best place to post this issue, as I have not found a hardware repository. 

When attempting to connect the 3.3V rail (or any 3.3V pin as HIGH) to the Vin of the TSOP38 (metal case), the output pin shows a 55.55Hz noise, every time, messing with the IR signal readings. 

I have tested the same circuit on a Wemos D1 Mini, which didn't output those noises. My thought is that there is happening an interference with Raspberry Pi 3 Model B (the only that I've tested so far) 3.3V Voltage Regulator. 

The solution that I came to my project was to use a voltage divider (100Ohm/100Ohm+100Ohm) to the 5V rail, and that worked, which confirms that the interference is originated in the 3.3V regulator + TSOP38 combination. 

The 5V rail directly to the TSOP38 did not output any background noise. The 3.3V rail directly to the Logic Analyzer didn't output any noise too. So it's a combination of both the 3.3V regulator and the TSOP. 

For newbies that don't have a solid knowledge of electronics, the solution of the voltage divider works good, since the supply current of TSOP is only 3mA.

lurch, 2017-06-04:
I'm very much an electronics newbie myself, but I wonder if a decoupling/smoothing capacitor would help?

arthurmoises, 2017-06-05:
Hi lurch! I've tried to to that too, before attempting to use the voltage divider resistors. I've used a 4.7uF film capacitor (those blue cubic ones) and a 680 Ohm resistor (which have output 608 Ohm on multimeter), to try to create a high-pass filter at the frequency of 55.55Hz. Even with those, I've managed to get some interference sometimes. 

I guess that I've could managed a better filter design, but as I was with lack of components at the time, I tough: "Why instead of trying to filter the noise, I haven't he noise at first?" So I discovered that the 5V rail didn't have the noise, and them, attempted to lower that to the voltage of 3.3V, which worked! For production board, I'll try to use an IC level shifter, instead of the voltage dividers.

JamesH65, 2017-06-06:
This is too off topic for the docs repo, so closing. I'd suggest using the forums for this sort of discussion.

closed 2017-06-06 by JamesH65

HFATL, 2018-05-03:
I've noticed also that my Raspberry Pi 3 is spewing much noise into my SDR radio which I listen remotely. As expected, any SDR on HF frequencies is very noise sensitive. Connected directed to a laptop the SDR radio gives much less noise.

In your experience the 3.3v regulator is the only culprit? I guess all I have to do is bypass the regulator and use a voltage dividir from 5V to 3.3V, correct?

Many thanks,

João Fonseca

JamesH65, 2018-05-03:
As I said above, this is NOT the place to discuss this, please use the forums, it will get a much larger audience.

N1MKG, 2018-05-03:
noise (#656)