raspberrypi/documentation #664: When Logged In

MarkP2015, 2017-06-08:
Please know that users reading this page, must be at least given a link to how to log in. It is never enough to simple say: "When logged in . . ."

Better still would be expanding this to include the ssh@xxx.xxx.xxx.xxx style commands.

bennuttall, 2017-06-08:
Which page are you referring to?

MarkP2015, 2017-06-08:
https://www.raspberrypi.org/documentation/linux/usage/users.md

JamesH65, 2017-06-19:
Do we really need instructions on how to log in? Seems rather simple/trivial to require a page. Happy to do one, just not convinced the need is there.

MarkP2015, 2017-06-19:
I'm a novice at Pi stuff. When I read the above, it seems to me that there is some instruction missing. I do not ask for a "page" as you put it, only complete instructions to provide the help that the reader can fully understand.

bennuttall, 2017-06-19:
The main Raspbian image logs you in (and boots to desktop) by default. You just open a terminal window and you're ready to enter the commands on that page.

Every page of the documentation requires you to be logged in, so it's surely unnecessary to state this at the top of each page, when it's the default anyway.

Unless I'm missing something - can you clarify the state of your Pi that leads you to be not logged in? Is it a very old image? Or Jessie Lite? Or not Raspbian?

MarkP2015, 2017-06-20:
I came to the page listed. I tried to follow the instructions. I could not log in. As for the suggestion that "Every page of the documentation requires you to be logged in . . . " addresses exactly what I'm asking after. I could not follow that pages instructions. This conversation started so long after my post, that my memory is now too uncertain of those events to continue. Thank you, Pi slicers for the "community". I'm out.

closed 2017-06-20 by MarkP2015

JamesH65, 2017-06-20:
Wow. So, we ask for a little more more information on what your problem
actually is (You seem unable to log in for some reason), and you go off in
a huff....

If you have an issue you should use the forums to get help. If there is a
problem with the documentation, then you should register an issue here.
This is the first time that help logging in has been flagged up here
(documentation), so after 13M sales, its not really surprising we want a
bit more information before we spend money changing the documentation.

Just for your benefit, here is how you log in if you are using a system
that actually requires you to log in.

Turn the device on.
Type your username (the default is pi)
Type the password (the default is raspberry)

You are now logged in. This process has not changed in the 5 years of the
Pi being on the market, or the 40 years that Linux/Unix has been in
existence. It's also very similar to the process in windows.

Does it really need describing in the docs?