raspberrypi/documentation #665: scp instructions incorrect

JamesH65, 2017-06-13:
Reported by email.

> I was trying to follow the information about file moving with scp at:
> 
> https://www.raspberrypi.org/documentation/remote-access/ssh/scp.md
> 
> The examples would not work for me. I find that I must enclose the file
> name in quotes. With that change, the example works.

XECDesign, 2017-06-13:
Any idea what the filename was? Unless it had spaces or something, it should work.

bennuttall, 2017-06-13:
The instructions are not incorrect. However, it's probably worth noting that files with spaces in them require quotes.

Filename with no spaces:

```bash
scp myfile.txt pi@192.168.1.3:
```

:+1:

File with spaces:

```bash
scp my file.txt pi@192.168.1.3:
```

:-1: 

Correct usage for filename with spaces:

```bash
scp "my file.txt" pi@192.168.1.3:
```

:+1: 

Spaces can also be used to separate files:

```bash
scp file1.txt file2.txt pi@192.168.1.3:
```

:+1:

JamesH65, 2017-06-13:
From the linked document

> To copy multiple files within a directory, you can use wildcards (e.g., * or ?). However, to use wildcards for copying multiple source files from a remote system, you need to place quotes (" ") around the path to the source files. This is necessary because the Unix shell, not the scp command, expands unquoted wildcards.
> Therefore, to copy all the .txt files from the revenge directory on your deathstar.com account to your revenge directory on empire.gov, enter:
> 
>  scp dvader@deathstar.com:"revenge/*.txt" ~/revenge/
> You'll be prompted for your password on the source system (deathstar.com). The command won't work unless you enter the correct password.

JamesH65, 2017-06-27:
I'm not sure what to do with this, perhaps just add the extra bits as Ben has suggested. I'll put it on the list.

bsimmo, 2017-07-07:
Why not just have it with quotes around, while it's two more characters to type, it should work for everyone with or without spaces or wildcards.

It's then not wrong, but perhaps over engineered in some settings.

commit 73a13149 (2017-10-17): Add note about spaces in filenames for scp command, close #665 (#734)

* Add note about spaces in filenames for scp command, close #665

* copy edits
closed 2017-10-17 by JamesH65